EdgeTX/edgetx #4570: Trimmer settings differ FM0 FM1

mrperno, 2024-01-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Different trim settings in different FM. FM0 and FM1 differ from the rest. 2.9.3 and dayly 2.10,0

### Expected Behavior

Same parameter in all FM

### Steps To Reproduce

Will show up during model setup

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

[image]
[image]

3djc, 2024-01-26:
FM0 is different by design as some of the options that exist for other FM are based on FM0, where FM0 doesn't have another FM to base on

pfeerick, 2024-01-26:
This is not a big or a mistake, but by design. As JC said, FM0 **is** different. It is the base/default FM from which all the other FMs derive from.

closed 2024-01-26 by pfeerick