EdgeTX/edgetx #4571: [CPN] 6POS Switch designation; GPS Input field in Simulator

ParkerEde, 2024-01-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

As far as I remember, we have already solved this before. It still works correctly in 2.9 Companion. In the current Main, the designation of the 6POS switch and also the GPS default value in the simulator is a
5x { xxxxx} e.g:
[image]
[image]

### Expected Behavior

fields should have the correct content

### Steps To Reproduce

1. start companion
2. start simulator
3. open telemetry simulator
4. observe the GPS field
or inspect the Radio Settings in companion from a radio with 6POS Switch

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS), RadioMaster TX16S / TX16SMK2

### OS Version

windows

### Anything else?

I've build the current main with german translation

pfeerick, 2024-01-27:
Thanks @ParkerEde ... it looks like #4445 somehow re-introduced that... the QtLinguist automatic parser must be misbehaving...

pfeerick, 2024-01-27:
Fixed in 6e37308

closed 2024-01-27 by pfeerick