EdgeTX/edgetx #4574: Frsky SRx series receivers Stab setup LUA script does not cover the second group of stabilized 5 channels in ACCESS mode

BompaBert, 2024-01-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Frsky added a second group of stabilized channels in ACCESS mode as from firmware version v1.0.5. So far, the latest LUA stab setup script does not let us set, enable or disable the second group of channels what potentially turns those channels unusable - the stab function for those channels is enabled by default.

### Describe the solution you'd like

Update the LUA script so it allows to control the second group of stabilized channels. The Ethos LUA script has another page that allows you to do just that.

### Additional context

LUA script version v2.04 and SR10+ firmware version v1.-.10       https://www.frsky-rc.com/archer-plus-sr10/

pfeerick, 2024-01-27:
You need to contact FrSky if they have added new features to the RX firmware, but not updated their Lua scripts to suit. This is not something EdgeTX can do or fix.

BompaBert, 2024-01-27:
I believe some in this community did create some great LUA scripts for
Frsky receivers. I was hoping to get their attention. Frsky is focusing on
Ethos - am still waiting for a reply.

closed 2024-01-27 by BompaBert

pfeerick, 2024-01-28:
That's why we need Frsky customers to complain (or compliment, where relevant), in order to get them to provide the necessary support... now that I look myself though, it seems like it is missing for the Taranis X9D+2019 also, yet is mentioned in the manual so should have been available as an option right from the start... so I will probably be emailing them about it also 😢. Offer has done a great job of enhancing the SRx Lua scripts provided by FrSky (for colorlcd) in https://github.com/EdgeTX/edgetx-sdcard/pull/113 (included in the 2.9.0 sd card pack onwards) - but depending on what Frsky did this may or may not work with the new capabilities, and may not be (easily) possible to be made to work without knowing what changes they may to the API to support this feature. 

Given it's CH1-6 and then CH7-11, probably "just" needs a second page?

BompaBert, 2024-01-28:
> scripts provided by FrSky (for colorlcd) in EdgeTX/edgetx-sdcard#113

To my knowledge, any radio not running Ethos wont be able to control the
SRx+ receivers second group of channels.  only the Ethos LUA script covers
the second stabilized group with a second page see also
https://www.youtube.com/watch?v=gzc1VwEPfCE - jump 13.44 mins further to
see both pages.

BompaBert, 2024-01-30:
@PFeerick, I got it working by adjusting the existing Frsky script (https://www.frsky-rc.com/archer-plus-sr10/). Everything on 1 page. Am glad I can now control everything. Not sure though if there is any infringement by grabbing their LUA and adjusting it.
Any help welcome in educating me as I do not mind publishing the script here.

pfeerick, 2024-02-01:
@offer-shmuely What do you think? ;)

offer-shmuely, 2024-02-01:
well let me try to answer it step by step
1. are you talking about SR10-Pro? or SR10_plus, by looking in the scripts, it looks like the firmware between the two is not compatible
2. are you sure that the script I post on edgeTx-sdcard is working on the plus receivers???
3. I do not have such receivers, so even if liked to, I can not test it.

I downloaded the setup scripts for both receivers
4. the name of the script of the old one, the pro, is FrSky_STAB_RX_2.03, and it is for openTx/edgeTx, on the new one, similar name FrSky_STAB_RX_v2.06 the ethos script, and the openTx script is now named AP_SRx_Lua_v2.04, if you can confuse the users, so why not ;-)
5. so if you have both SR10-Pro & SR10-plus, what script do you put on your transmitter? it is not clear at all
6. the script for the pro & the script for the plus, use different command codes, so they does not look compatible at all, did you note that on the video you send, he could not operate the script!!!
7. so it is a total mess. for the plus it will need a totally new script  
8. this is why I named the script S6R_S8R, so there will be no misunderstood

pfeerick, 2024-02-02:
[As per the link](https://www.frsky-rc.com/archer-plus-sr10/), this is in relation to the SR10+... the SR10 Pro is a completely different beast (and uses STAB_RX_2.03 -> FrSky SxR_R9S lua to configure it). And yes, just a few things confusing about this whole thing... like the need to downgrade the receiver and use older Lua script for the initial calibration :-/ 

The underlying issue is that you use the AP_STAB_RX_v2.04 (AP Stab Rx_OPTX, AP Stab Rx Cali_OPTX) scripts as provided by Frsky to configure it on OpenTX/EdgeTX, but the second group of stabilisation options mentioned in the manual are not available to be configured in the OTX script, only the Ethos one. Yet it seems like there is only a *minor* modification needed to make it all work... https://github.com/EdgeTX/edgetx/issues/4574#issuecomment-1913589133

I can't say if the S6R_S8R script works for the SR10+ as only I use X9D+ with ACCESS ;)

BompaBert, 2024-02-02:
> Confirmed. This is for the SRx+ series only.
> Attached is a modified working script. I used the FrSky OTX / Ethos script
> version 2.04 to do this. It has been tested on a SR10+.
> I do not have any of the other models to test it with but albeit not the
> latest version for some, the v2.04 script still shows on their website.
> The main and second group of stab channels are all on 1 page. My skills
> don't go that far to split this on 2 pages like Frsky and Offer did.
> What you think?

Also confirmed that - with latest firmware -  the SRx PRO script does not
work for SRx PLUS receivers and vice versa.
The latest script (v2.06) added roll/pitch angle. I'll test that soon.

BompaBert, 2024-02-02:
Your  S6R_S8R script only works for ACCST receivers. Wont work for ACCESS -
SRx PRO or SRX PLUS.

offer-shmuely, 2024-02-02:
If i had the receivers, I could try to add it, at least for the pro.

It is very strange!! that the script that we put on the edgeTx sd card is working with the plus receiver, i did not expect it will.

BompaBert, 2024-02-02:
Just very confusing indeed. The SxR series are ACCST only. Then the SRx
PRO's  uses ACCESS. Both are discontinued and seems to work with your
script or the FrSky SxR_R9S / FrSky_STAB_RX_2.03 scripts. The SRx PLUS does
not. It would be great if you could take a look at what was attached
earlier - make it nicer and maybe publish?

offer-shmuely, 2024-02-03:
I saw the attached script you sent, and the original sr10+ scripts, they both use different command codes than my script.
Are **sure** my script ever worked on the SR10+?
That would be very surprising

offer-shmuely, 2024-02-03:
@BompaBert, can we move to discord for that discussion?

BompaBert, 2024-02-03:
Yours does not work for the plus series. Sent from my iPhone

offer-shmuely, 2024-02-03:
try this code
I remind you, that I have no way to test it, I do NOT have this receivers!!

[FrSky_SR10_plus_setup_v2.05-etx-SR+.zip](https://github.com/EdgeTX/edgetx/files/14154542/FrSky_SR10_plus_setup_v2.05-etx-SR%2B.zip)

GerritErpenstein, 2024-02-04:
> try this code I remind you, that I have no way to test it, I do NOT have this receivers!!
> 
> [FrSky_SR10_plus_setup_v2.05-etx-SR+.zip](https://github.com/EdgeTX/edgetx/files/14154542/FrSky_SR10_plus_setup_v2.05-etx-SR%2B.zip)

I have tested the script with an Archer Plus SR8. It gave me the following error:
```
Script syntax error
TOOLS/FrSky_SR10_plus/setup.lua:327: bad argument #4 to 'sportTelemetryPush' (number expected, got nil)
```

BompaBert, 2024-02-04:
Indeed -syntax error.
@offer, you can test this using the etx companion - grab a model and run a
simulation...

BompaBert, 2024-02-04:
@Gerrit, could you test attached. It works on the SR10+. It would be good
to try the SR8+

GerritErpenstein, 2024-02-04:
> @gerrit, could you test attached. It works on the SR10+. It would be good to try the SR8+

I would love to test the script, but I can't find it here in the GitHub web UI. Seems like it is only available via E-Mail? It would be great if you could upload it here again.🙂

BompaBert, 2024-02-04:
Here you go.

GerritErpenstein, 2024-02-04:
Still no script attached. Please see the original issue: [4574](https://github.com/EdgeTX/edgetx/issues/4574)

BompaBert, 2024-02-04:
here you go. Forgot it does not work via email.  [SRxPLUS_setup.zip](https://github.com/EdgeTX/edgetx/files/14157953/SRxPLUS_setup.zip)

offer-shmuely, 2024-02-04:
I do use the companion, 
I added some mock so I can see the screens, but anything that is actual read/write to the RX is not working.
anyway, try this one, maybe I could guess the bug

[FrSky_SR10_plus_setup_v2.05b-etx-SR+.zip](https://github.com/EdgeTX/edgetx/files/14158353/FrSky_SR10_plus_setup_v2.05b-etx-SR%2B.zip)

GerritErpenstein, 2024-02-04:
Crap, I just realized that the configuration of the second gyro page only works with ACCESS. However, I use ACCST. I should have read the title more carefully.
This makes the receiver with ACCST practically useless. What was FrSky thinking? 😞 Then I will probably send the receiver back again.

offer-shmuely, 2024-02-04:
could you understand why, when using the 2 group, it have 2 mount types?
and 2 plane-type?
how can you put a single receiver in two different orientation?

pfeerick, 2024-02-04:
You would have to ask frsky that, as they clearly know that is the case as it is outlined in the setup steps in the manual 🤣

Given the second stabilisation group when turned on controls CH7-11, maybe it's so that you can swap it between planes? As the second group also has independent calibration. I can't offhand think of any other reason why you would want duplicated controls with different control mix/orientation, unless maybe for some form of camera stabilisation through inversion?

BompaBert, 2024-02-04:
@offer, you nailed it. It's working! Feel free to add it to the etx sdcard.
Just we all agree, this version does not cover the latest features. As
mentioned, FrSky recently added v2.06 -  are roll/ pitch angle. I tried to
add them but they return no values on the sr10+ I tested it with.

BompaBert, 2024-02-04:
The second group has its own self-check, not calibration as far I know. A
second group can be useful in more complex setups like jets with tailerons.
One group stabilizes the ailerons and a second group stabilizes the
elevators in a delta setup (ail+ele).

pfeerick, 2024-02-04:
It actually says calibration in the manual, but it probably doesn't mean that (or our interpretation of calibration is wrong).

BompaBert, 2024-02-04:
Yes sir, confusing indeed. the calibration happens using a different script.

3djc, 2024-02-04:
The calibration for sxr allows to do initial calibration of accelerometers. It should be done once in receiver lifetime, and before installation in a craft, since it need flat leveled reference surface. It is not part of receiver config per say

pfeerick, 2024-02-04:
Actually, no it isn't. You have calibration of the IMU "gyro", which is done in the seperate script, and appears to be one time (unless it needs resetting), and calibration of channel travel/centering, so the stabiliser knows how each bank is set up. That's the self-check calibration.

3djc, 2024-02-04:
Oh  yes, the channels calibration, yes, there are two different ones. One should be renamed really, but since frsky docs are what they are , not surer what can be done

offer-shmuely, 2024-02-04:
the script have different command-id for each of the wing-type & and plane-type
it is not a mistake in the documentation

[image]

offer-shmuely, 2024-02-04:
try this for the roll/ pitch angle
[FrSky_SR10_plus_setup_v2.05c-etx-SR+.zip](https://github.com/EdgeTX/edgetx/files/14158675/FrSky_SR10_plus_setup_v2.05c-etx-SR%2B.zip)

BompaBert, 2024-02-04:
The roll/pitch values can be altered from 0 to 80 however I do not see any
change in servo movement. Followed instructions per TD R10 manual: " This
feature will not be activated if the degree value is set to 0-9. The
setting with the Roll/Pitch degree between 10 to 80 can have the feature
work, and it's not recommended to be set with the degree values too small."
So it says "can" - makes me unsure...
Also on 3th page, self check can't be selected and am getting intermitted
crash with message:
?.0: attempt to index field '?' (a nil value)

pfeerick, 2024-02-04:
> the script have different command-id for each of the wing-type & and plane-type
> it is not a mistake in the documentation

Never said it was a mistake, but that would need to ask frsky about the rationale... Then again, they also clearly state these are two separate stabilisation systems, so it stands to reason everything is just duplicated. 

But as Boompa pointed out, it starts to makes sense when using more complex setups, I.e. the second bank for delta mixing (with no mixing for the first bank).

BompaBert, 2024-02-04:
Lets hope for the best. So far, a lot of progress was made. Thanks much for
that.
Btw, on page 3  of your script, you've got a typo. Ch10 represents AIL4 and
is still controlled by the gyro. Am referring to the sr10+ manual.

offer-shmuely, 2024-02-05:
I do not understand the mistake
> on page 3  of your script, you've got a typo. Ch10 represents AIL4 and is still controlled by the gyro

as far as I understand, if you like the CH10 to be controlled by the gyro as AIL4, you turn this ON
[image]

if you like to use CH10 for something, lets say "gears" then you need to turn it OFF, same as CH5/6

 Roll/Pitch degre
I see it only on the manual of the SR6+, not on the SR8+/SR10+/SR12+

> attempt to index field '?' (a nil value)

for that, I need you to delete from the sd card the luac file \SCRIPTS\TOOLS\FrSky_SR10_plus\setup.luac 
not the lua itself, run it again, and we will see a m ore detailed error.

pfeerick, 2024-02-05:
The manual is out out date somewhat (and is itself inconsistent... it is v1.0, yet, mentions something about the 1.07 firmware :thinking: )... this is from the firmware log for the SR10+:

[image]

BompaBert, 2024-02-05:
No errors since removing the .luac file. In regards CH 5-6 and 10-11. Your comment just confused me -sorry about that. Reason is that FrSky calls it "AUX" mode since the time they came out with stabilized receivers. In attached, I made a few changes in regards to that, fixed the self-check choice and uncommented the roll/pitch feature in group2. Albeit I do not see any changes in roll/pitch behavior changing the degrees on the bench but everything else works. I'll test the roll/pitch behavior in flight - maybe it is acceleration depended.
[setup_v2.zip](https://github.com/EdgeTX/edgetx/files/14167265/setup_v2.zip)
...

offer-shmuely, 2024-02-05:
there is no real meaning tor AUX.
openTx & edgeTx does not name the channel as AUX (and good that they don't!)
on betaflight AUX1 is actually CH5
on rotor-flight AUx1 is actually CH6 (since there is a pitch as well in addition to AIL/ELE/THR/RUD)
I do not see any reason to continue and use the AUX naming

e.g.
if I like to config lights on AUX11, what channel I need to use on the transmitter? 
on betaflight it will be 11+4=CH15
on rotorflight it will be 11+5=CH16
and after configure it, there still will be no reference for that in edgeTx at all

if the first text is confusing, lets find another one that not contain AUX

BompaBert, 2024-02-05:
What about       { "AIL2", "stab off" }

pfeerick, 2024-02-05:
> on betaflight AUX1 is actually CH5
> on rotor-flight AUx1 is actually CH6 (since there is a pitch as well in addition to AIL/ELE/THR/RUD)

... except we're talking about Frsky, not Betaflight or Rotorflight... what language/phraseology does Frsky use? As this is what the end user will expect. The same as I  would (and do) expect Betaflight scripts/tools to reference AUX1 as CH5... that's the language of *that* system.

Also, consider the option(s) in context... 
```
CH5  mode = [AIL2|AUX]
CH6  mode = [ELE2|AUX]
CH10 mode = [AIL4|AUX]
CH11 mode = [ELE4|AUX]
```
it seems pretty clear that, for example, CH10 is either being used as AIL4, or as an auxiliary channel, not a *specific* channel. I'm not sure there is any benefit in expanding the option text out to "CH10 as AIL4" since we're only looking at CH10... and only has those two options. If you were remapping channels, sure, but not for what amounts to an stablise/user control option.

Maybe something more like this if you want to make it more explanatory?
`{ "AIL4", "AUX (Not stabilised)" }`

I was going to suggest
`{ "AIL4 (Stabilised)", "AUX (Not stabilised)" }`
but I'm not sure what happens when you turn one of the stabiliser groups off... i.e. I would expect the channel assignment to remain, but the channel to not be stabilised, so that text would be wrong in that instance.

offer-shmuely, 2024-02-06:
The "AUX (Not stabilized)" } sound good to me (as long as the AUX does not have number) 

changing the ON string to "stabilized as AIL2"

[FrSky_SR10_plus_setup_v2.05d-etx-SR+.zip](https://github.com/EdgeTX/edgetx/files/14173268/FrSky_SR10_plus_setup_v2.05d-etx-SR%2B.zip)

BompaBert, 2024-02-06:
I have been using the FrSky receivers sine they came out. Being used to
AUX1 AUX2 etc...so initially, Offer's description throwed me off.
Personally I would not add any comments and stick with the Frsky way of
doing things. FrSky receivers can be complex to handle and  if you don't
want to get in trouble, reading / understanding the available manuals and
release notes is a must and tools like this one should not replace or
change that. Adding stabilization comments or descriptions in a
stabilization tool is in my opinion overkill. And a Peter mentioned,
disabling a group removes all stabilization in that group. All do respect,
I would go with the FrSky way or what Peter suggested:

CH5  mode = [AIL2|AUX]
CH6  mode = [ELE2|AUX]
CH10 mode = [AIL4|AUX]
CH11 mode = [ELE4|AUX]

3djc, 2024-02-06:
That will also help consistency with BW

BompaBert, 2024-02-07:
The AIL stab functionality fails when the receiver is mounted vertical or vertical reverse. It works ok horizontally or reverse mounted. Rolling a bit to one side, lets say the left, throws the servo to max. Rolling to the right, stabilization is ok. Will update on further testing and findings.

offer-shmuely, 2024-02-09:
can you compare it to the original script?

BompaBert, 2024-02-12:
Here is my version. Am using 2 pages. On page for each stabilization group. You can switch the pages back and forth. Everything seems to work on the SR10+ with v1.0.10 firmware except the roll and pitch angle. Am not 100% sure the roll/pitch angle script values are ok neither do I have a clue what it supposed to do. Playing with different angle values does not changes anything, at least not on the bench. Any comments welcome.
[SRxPLUS_setup.zip](https://github.com/EdgeTX/edgetx/files/14250808/SRxPLUS_setup.zip)

Ralf-W, 2024-02-13:
> Here is my version. Am using 2 pages. On page for each stabilization group. You can switch the pages back and forth. Everything seems to work on the SR10+ with v1.0.10 firmware except the roll and [...]

Thanks a lot for your significantly improved LUA script "SRxPLUS_setup.zip". I installed it into my TX16S, running EdgeTX 2.9.3, using ACCST mode, FrSkyX2 LBT(EU), MPM Version v1.3.3.33.
Page 1/2 now allows to input those roll and pitch limit values - and I can report that this feature does work correctly:

As an example I have set the roll limit to 60 degrees. For the bench test, I keep the SR10+ receiver horizontally and do not move it, while I control the aileron stick of my Tx to +100%. This moves the roll servo nearly into one end position. Now I tilt the Rx slowly into the desired roll direction. This reduces the deflection of the roll servo continuously, reaching it's neutral position exactly when the telemetry displays a roll ange of 60°. 
In flight the full +100% aileron stick command will make the airplane roll into the desired direction - but the stabilizer mode of the Rx will reduce the aileron deflection continuously until the model has reached a bank angle of 60 degrees. Then the aileron servo is back to its neutral position, where it will be kept and stabilized by the Rx - though the aileron stick at the Tx is still fully deflected to +100%.

I tried different settings for the target roll angle, keeping the Rx fixed to the bench: As stated in the manual, there is no bank angle limiting between 0° and 9°. For small bank angle limits (beginning at 10°) the maximum servo movement that can manually be controled by the aileron stick is significantly reduced. It nearly reaches full servo deflection for the maximum allowed bank angle of 80 degrees. In flight the Rx will be rotated together with the airplane and the stabilization will reduce the manually controled  servo deflection immediately until the target bank angle is reached.

Thus, your improved LUA script activates the bank angle limit feature for me. 

However, I do see the second page with all the parameters for the second "Stable System 2" - but the values there don't get populated. Therefore I cannot switch off the stabilizing for channels 7-11.
Yes - I updated the Rx to firmware version V.1.0.10. Nevertheless, some of my settings are not correct.

What are your successful firmware selections (EdgeTx , multimodule , Rx) to be able to readout and change the values of "Stable System 2"?

BompaBert, 2024-02-13:
@Ralf-W , you will need to run in ACCESS mode to get the second stab group to work -thanks for the testing though. My setup is the Frsky Horus X10S Express, running EdgeTX 2.9.3 with a SR10+.
Am attaching the complete directory here -it includes the calibration script using custom pictures -the Frsky script can be confusing. And comments are welcome.
[SRx_PLUS_v2.05.zip](https://github.com/EdgeTX/edgetx/files/14268523/SRx_PLUS_v2.05.zip)

BompaBert, 2024-02-20:
FYI. Tested roll/pitch angle in flight and it works fine - in auto level mode the roll/pitch angle can be reduced by settings the degrees in the setup script from 10 - 80. No limitation when leaving the setting to 0 - 9 degrees.

crogersqx7, 2024-03-29:
Has this working script with pitch and roll angle been added to the sd card contents??

offer-shmuely, 2024-03-29:
No, it is not
There is no one that can really support it

offer-shmuely, 2024-03-29:
You can find the last version on my repo, but I cannot support it, since i do not have the hardware

BompaBert, 2024-03-29:
[SRx_PLUS_v2.05.zip](https://github.com/EdgeTX/edgetx/files/14268523/SRx_PLUS_v2.05.zip) is working fine. Pitch, Roll and both stabilization groups- tested on the SRX series of receivers. It has been made public via RCGroups and so far, no issues. This is my version though but feel free to take a look and eventually add to the SD card.

BompaBert, 2024-04-08:
@pfeerick , FYI - above script is partially failing with EdgeTX version 2.10 (Hedgehog). The second page can't be chosen anymore.

pfeerick, 2024-04-09:
What radio? I can't test on a colorlcd Frsky radio (only have the X9D series radios), but the TX16S is able to page back and forwards between the two pages in SRxPLUS_setup (with 2.10.0-rc3). If you're still running RC1 or RC2, you probably need to update to RC3 as some missing EVT_PAGEDN_ events were re-added, which is probably why it isn't working for you.

[8 more replies]