EdgeTX/edgetx #4580: FPV simulator DCL - The Game on PS4 Pro does not detected Radiomaster Boxer when connected.

YeTuMan, 2024-01-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

FPV simulator DCL - The Game on PS4 Pro does not see Radiomaster Boxer EdgeTX 2.9.0 - 2.9.3 when connected, on EdgeTX 2.8.5 works fine and is detected

### Expected Behavior

Radiomaster Boxer EdgeTX 2.9.3 should be detected when connecting it to the console.

### Steps To Reproduce

1. Start simulator DCL - The Game on PS4 Pro
2. On Radiomaster Boxer
3. Connect USB C to USB to PS4 Pro and Radiomaster Boxer
4. On Radiomaster Boxer select Joystick 
5. Go to menu settings
6. Go to menu controls
7. Go to menu transmitter calibration
8. Select refresh devicer

### Version

2.9.3

### Transmitter

RadioMaster Boxer

### OS Version

PS4 Pro

### Anything else?

On Radiomaster Boxer EdgeTX 2.8.5  works fine and is detected
PS4 Pro version firmware 11.02
USB C to USB  cable

pfeerick, 2024-01-28:
Have you tried any advanced interface modes (new in 2.9) such as gamepad and multi-axis? Please note that when using the advanced usb joystick mode you *will* need to then go into channel settings, and tell it what each channel is to be... i.e. Axis, button, etc.

https://edgetx.gitbook.io/edgetx-user-manual/b-and-w-radios/model-select/setup#usb-joystick

YeTuMan, 2024-01-28:
Yes, not detected
I have on PS4 Pro Liftoff and DRL Simulator - work fine in Classic mode.

YeTuMan, 2024-01-28:
It feels like I didn't connect it to ps4 pro/

iuqdwnjk, 2024-01-28:
Same issue on ps5, also tried all advanced modes, the device selection in DCL shows only “None”,
tried with my Jumper T-Pro v2, latest stable fw and a recent nightly build.

ThomasKuehne, 2024-02-11:
I had a similar issue with an other radio controller.
A USB-C to USB-C connection only charged the device, using any USB-A cable in the connection actually enabled Joystick and SD-card access.

pfeerick, 2024-02-11:
That is probably the only suggestion I could make also... double check with
another device if that cable is actually a data cable or charge only
(charge only should trigger the prompt on the radio, but not actually
work). Orientation of the connector "may" matter also, so also try flipping
the connector. Otherwise it sounds like a weird compatibility issue.

ThomasKuehne, 2024-02-12:
YeTuMan:
-  Pro Liftoff and DRL Simulator do detect the Boxer 2.9.0 and treat it as a joystick
-  using the same EdgeTx model, cabels and classic Joystick mode as in 2) the FPV simulator DCL doesn't detect the joystick
-  FPV simulator DCL however works if you downgrade to 2.8.5

iuqdwnjk:
-  FPV simulator DCL also doesn't work with 2.9.3 on a Jumper T-Pro v2

Questions:
- [ ] What is the exact name / homepage of the simulator? Google finds only this ticket.
- [ ] Does FPV simulator DCL work if you set channel 8 in the mixer to -100% ? (see Dial axis below)
- [ ] Does FPV simulator DCL detect a 2.9* with following advanced joystick configuration? 
- mode: Gamepad
- CH1: axis X
- CH2: axis Y
- CH3: axis Z
- CH4: axis rot**Z** (yes Z, not X)
- CH5-8: empty
- CH9-14: button 0-14
- 13+: empty
- [ ] if that doesn't work: unplug, change mode to Advanced / Joystick and try again 
- [ ] If you are adventurous: test with pull request #4626

A USB hid session on Linux (PS4 debugging is a bit more challenging) doesn't reveal any obvious clues:

```
2.8.5 HID desciptor dump
 05 01 09 05 A1 01 A1 00 05 09 19 01 29 18 15 00
 25 01 95 18 75 01 81 02 05 01 09 30 09 31 09 32
 09 33 09 34 09 35 09 36 09 36 16 00 00 26 FF 07
 75 10 95 08 81 02 C0 C0

2.9.0 HID desciptor dump
 05 01 09 05 A1 01 A1 00 05 09 19 01 29 18 15 00
 25 01 95 18 75 01 81 02 05 01 09 30 09 31 09 32
 09 33 09 34 09 35 09 36 09 37 16 00 00 26 FF 07
 75 10 95 08 81 02 C0 C0
```
The HID descriptors are different:
2.8.5 reports axis: X, Y, Z, Rx, Ry, Rz, Slider, Slider
2.9.0 reports axis: X, Y, Z, Rx, Ry, Rz, Slider, Dial

Would be surprising if the Dial is causing issues.

```
2.8.5 HID report dump
 03 00 00 FF 07 FF 07 FF 07 FF 07 00 04 00 04 00
 04 00 04

2.9.0 HID report dump
 03 00 00 FF 07 FF 07 FF 07 FF 07 00 04 00 04 00
 04 00 04
```

The HID report dumps are identical.
axis 5-8 have no mixer, as expected they report 1024 = 0%

ThomasKuehne, 2024-02-12:
An other sim hardcoded the axis asignments, the solution there was following trick

- [ ] Does FPV simulator DCL detect a 2.9* with following advanced joystick configuration?
mode: Gamepad
CH1: axis X
CH2: axis Y
CH3: axis Z
CH4: axis rotX
CH5: axis rotY
CH6: axis rotZ
CH7: axis slider
CH8: axis slider (yes, two sliders)
CH9+: fill button IDs 0 - 23 / Postion "8POS" instead of "Push" is usefull to claim dummy buttons

No need to actually configure input/mixer for CH5-8. The joystick simply has to claim those axis in that order.

ThomasKuehne, 2024-02-14:
The only other USB related issue I can think of is that DLC uses the manufacturer string from the USB to do some kind of matching.

In 2.8.5 EdgeTX reports:
```
  idVendor           0x1209
  idProduct          0x4f54
  iManufacturer           OpenTX

```  

In 2.9.0 EdgeTX reports:
```
  idVendor           0x1209
  idProduct          0x4f54 
  iManufacturer           EdgeTX
```
That was this change: [radio/src/targets/common/arm/stm32/usbd_desc.c](https://github.com/EdgeTX/edgetx/commit/5acaa79c1c18df88ae2d900af578dadec0153f4e#diff-3c88b6d11382b9df01b217797c34c1e24ef36bd9d97508552d0114f57e522f0c)

I've asked for more information on DCL's Discord server. 
Seems there was an informal process in the past https://dcl.aero/1-03-patch-on-ps4/ to add controllers.

devova, 2024-02-15:
The same issue on MacOS. Neither Liftoff nor https://hardwaretester.com/gamepad can't detect newly upgraded Radiomaster Boxer EdgeTX 2.9.3. Worked well on EdgeTX 2.8.5.
Actully Liftoff detects it as Boxer, perhaps via usb descriptor bot non of the channels are passed through.

pfeerick, 2024-02-15:
You have me curious now... I don't have DCL (yet, waiting for next sales) ... so am wondering if it is that... picky... as to also be checking the manufacturer rather than just the PID/VID... see if this works with the boxer and DCL... it's a build of the 2.9 (so basically 2.9.4) and I merely changed the manufacturer name used for USB mode back to OpenTX... 
[boxer-report-as-opentx-77884b6a2.bin.zip](https://github.com/EdgeTX/edgetx/files/14303228/boxer-report-as-opentx-77884b6a2.bin.zip)

devova, 2024-02-16:
@pfeerick I flashed your build, nothing has changed, the issue remains as it was.

iuqdwnjk, 2024-02-16:
I confirm, that my **Jumper T-PRO v2** works well in DCL with today's master build with changed manufacturer to `OpenTX` in `radio/src/targets/common/arm/stm32/usbd_desc.c`

I've also rolled back the previous version I've used and it stopped working, upgraded again to my new build and it's well detected again.

You can test it yourself again @devova 

[boxer-3ebff87e8.bin.zip](https://github.com/EdgeTX/edgetx/files/14310756/boxer-3ebff87e8.bin.zip)
[tprov2-3ebff87e8.bin.zip](https://github.com/EdgeTX/edgetx/files/14310757/tprov2-3ebff87e8.bin.zip)

3djc, 2024-02-16:
Rings a bell, I had to request an id for otx, way way back. Maybe there is matching done, will dig my archives

YeTuMan, 2024-02-16:
I too confirm that my Radiomaster Boxer works well in DCL and it's detected on my PS4 Pro with the **"boxer-3ebff87e8.bin"** build.

devova, 2024-02-16:
Unfortunately, I **can't** confirm that [boxer-3ebff87e8.bin.zip](https://github.com/EdgeTX/edgetx/files/14310756/boxer-3ebff87e8.bin.zip) works for me. Still same symptoms. A few debug info
```
1209 4F54: OpenTX - Radiomaster Boxer Joystick
DESCRIPTOR:
  05  01  09  04  a1  01  a1  00  05  09  19  01  29  01  15  00
  25  01  95  20  75  01  81  02  c0  c0
  (26 bytes)
```
[image]

devova, 2024-02-16:
So I decided to do a regression testing and find out on which version it actually become broken, being on custom [boxer-3ebff87e8.bin.zip](https://github.com/EdgeTX/edgetx/files/14310756/boxer-3ebff87e8.bin.zip) which was not working for me I flashed:
- v2.9.3 - doesn't work
- v2.8.5 - works great!
- v2.9.1 - works great
- v2.9.2 - works great
- v2.9.3 - works great (be aware that it wasn't working before)
- v2.9.4 - finally official build also works

```
# v2.9.4
> ./mac-hid-dump
...
1209 4F54: EdgeTX - Radiomaster Boxer Joystick
DESCRIPTOR:
  05  01  09  05  a1  01  a1  00  05  09  19  01  29  18  15  00
  25  01  95  18  75  01  81  02  05  01  09  30  09  31  09  32
  09  33  09  34  09  35  09  36  09  37  16  00  00  26  ff  07
  75  10  95  08  81  02  c0  c0
  (56 bytes)
```

So the were some problems in migration, because originally I was migrating from 2.8.5 to 2.9.4

YeTuMan, 2024-02-16:
> I wouldn't recognie that HID descriptor as a Joystick either - it delivers
> only one button and no axis.
>
> @devova <https://github.com/devova> did you use an advanced Joystick
> setup or the classic mode?

Use classic mode

devova, 2024-02-16:
Don't even sure how to use classic mode? I just connect Boxer via USB and select HID

ThomasKuehne, 2024-02-16:
Just for future reference: The "classic" or "advanced" joystick mode is the last entry in the model setup.

[image]

YeTuMan, 2024-02-16:
I understand this mode with the default settings specified in the manual

https://edgetx.gitbook.io/edgetx-user-manual/edgetx-user-manual/user-manual-for-color-screen-radios/model-settings/model-setup/usb-joystick

closed 2024-02-17 by pfeerick

HereInMyOwnSkin, 2024-04-01:
Sorry, I’m not sure if I’m supposed to open a new bug, but I’m having the same issue connecting a Taranis xd9+ to DCL The Game on PS4 Pro. It appears to connect without issue to PC, so I don’t think the hardware is the issue. I have recently moved from OpenTX to EdgeTX too because it stopped working. I have 2.9.4 firmware installed.

pfeerick, 2024-04-01:
You need to update to 2.10 when it releases, or use one of the release candidate builds then. This issue was closed as it was resolved by the PR referenced immediately above your comment ;)

HereInMyOwnSkin, 2024-04-01:
Thanks. I'll try that. I wasn't sure if 2.10 included the fix or if devova had suggested something had gone wrong in his original migration. Will try updating to 2.20 :)

HereInMyOwnSkin, 2024-04-02:
@pfeerick that worked thank you. It's connected, but unfortunately none of the inputs are working. Half way there!

HereInMyOwnSkin, 2024-04-05:
> @HereInMyOwnSkin 
> 
> > It's connected, but unfortunately none of the inputs are working.
> 
> 
> 
> In Model / Setup: What is your USB Joystick Mode? Should be `Classic ` for DCL.
> 
> 
> 
> Do the values change in the channel monitor when you move the sticks?
> 
> 
> 
> 

Yes, it only detects it when it's in Classic. Any advanced setting doesn't even detect the radio. All the channel monitors on the radio show changing values, just nothing on the game. Not even in calibration. I've turned off Internal and External RF too. Is that correct?

ThomasKuehne, 2024-04-05:
@HereInMyOwnSkin 

Yes turning off Internal and External RF is best practice and doesn't influence how games deal with the joystick.

From your description it sounds like DCL uses the [reverse axis order](https://manual.edgetx.org/edgetx-how-to/configure-advanced-joystick-with-edgetx)  to identfy axis.

Try following advanced configuration as a starting point

1. axis X
2. axis Y
3. axis Z
4. axis rotX
5. axis rotY
6. axis rotZ
7. axis Slider
8. axis Dial
9. button 0
10. button 1
11. button  2-9
12. button 10-17
13. button 18-23

If that works try switch axis X and axis Y. (first Y and then X). 
If the input from CH1 is still displayed in the same position in DCL, then DCL identifies axis by their order.

Then remove axis Dial. If DCL stops working or shifts the inputs then it is using the revers axis order. Meaning you'll have to define 8 axis even if only 4 are actually used.

HereInMyOwnSkin, 2024-04-07:
@ThomasKuehne sorry, I'm not sure what you mean. I have the radio set up as TAER (I think!?) and DCL seems to suggest it's looking for that. Do you mean changing from classic to advanced? Thank you!

pfeerick, 2024-04-07:
Yes, he is talking about setting it up in advanced mode. When you have it
configured for advanced joystick mode, yoiu'll be able to configure each
axis as described, in order to see if that is what DCL is looking for.

When you said "Yes, it only detects it when it's in Classic." before... do
you actually mean "it is detected, but none of the sticks work, when it is
in Classic mode"?

HereInMyOwnSkin, 2024-04-08:
@pfeerick yes that's accurate. In classic it detects the radio, but none of the controls work. I shall try advanced later today. Thanks both.

HereInMyOwnSkin, 2024-04-08:
@ThomasKuehne I added those all in advanced. It still didn't detect unless I switched back to Classic, but it's picked them all up now. Swapping X and Y didn't seem to have any effect, nor removing Dial. My only issues seems to be that the throttle only works from half throttle to max, but is -ve to +ve on the radio. Thanks again for you help (and @pfeerick)!

HereInMyOwnSkin, 2024-04-15:
Well I thought i understood what was going on, but I can't work it out. It only seems to connect with Classic. It then detects the following. Twice now I've managed to get it to work, but I can't actually work out what it is that's making that happen. Initially it just seems to detect as in the picture, but none of the inputs actually do anything. [image]

LucasD11, 2025-04-14:
Still have the same issue with 2.11.0-rc3. With classic or advanced as described here - https://github.com/EdgeTX/edgetx/issues/4580#issuecomment-2040541613

PR #4653 (2024-02-17): fix(radio): "DCL - The Game" not detecting USB joystick
stop gap change to fix #4580 by reverting the USB manufacturer string to the pre-2.9.0 state ("OpenTX" instead of "EdgeTX")