EdgeTX/edgetx #4586: Voltage sensor minimum tracks sensor maximum when sensor reaches new maximum value

pagrey, 2024-01-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

A sensor, A1 or RxBt, A1- resets to A1+ when A1 is greater than a previously stored maximum value. Same result on simulator and the radio itself with a connected RX.

### Expected Behavior

I expect A1- to retain a low value until telemetry is reset.

### Steps To Reproduce

Create a telemetry screen with A1, A1- and A1+. Simulate A1 or connect a RX with A1 and watch A1- as A1 changes. I have tested on the simulator.  FrSky, DSMX and FlySky have similar results with RxBt.

### Version

2.9.3

### Transmitter

RadioMaster Boxer

### Operating System (OS)

Linux

### OS Version

Arch 6.7

3djc, 2024-01-31:
Confirmed in 2.10

3djc, 2024-01-31:
Ok, here is what happens, that behavior dates Otx in 2016.

The principle here is that voltage cannot (should not) increase over time, but always decrease as your batt depletes. If voltage rises about RxBt+, the radio assumes you did put an new batt in the craft, and therefore RxBt- is updated to reflect the new minimum if THIS battery.

pagrey, 2024-01-31:
That explains it, thanks for the reference to the code as well, that really helps.

closed 2024-01-31 by pagrey