EdgeTX/edgetx #4590: TX16S simulator writes input Trim sources as None

elecpower, 2024-01-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

Simulator displays the values correctly on loading however sets all trims as input to None on writing

### Expected Behavior

Trims written as set

### Steps To Reproduce

1. Create a new model in Companion
2. Set separate input lines sources to TrmR, TrmE, TrmT, TrmA, Trm5, Trm6
3. Save the file
4. Start the standalone simulator and select the newly created file
5. Go to Model Inputs and values as set in Companion
6. Exit the sumulator
7. Start the simulator as before
8. Go to Model Inputs and sources are set to ----

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Linux

### OS Version

Ubuntu 20.04, Win 10

### Anything else?

Commit 0a2cf011

Companion
[image]

Simulator 1st run
[image]

Model file after 1st run
[image]

Simulator 2nd run
[image]

philmoz, 2024-01-31:
I think this is a Companion issues.

For example if the source is set to Rud Trim then Companion writes 'srcRaw: TrimRud'.
After the ADC refactor PR, the firmware writes 'srcRaw: T1' which Companion does not currently handle when reading the model file.

elecpower, 2024-01-31:
@philmoz thanks for the info. I'll recheck my steps. I've gone round the testing loops so many times it's likely I've blundered.

elecpower, 2024-02-01:
@philmoz the bug is me. I forgot the simulator close writes from the temp storage back to the source file via Companion storage and thus why the file is broken post simulator exit The temp file is as you stated contains T1 - Tn.

closed 2024-02-01 by elecpower