EdgeTX/edgetx #4593: Increase precision of Slow Up/Slow Down for surface controllers

kmf123kmf, 2024-02-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Mixer Slow Up and Slow Down precision of 100ms is insufficient for surface vehicle steering and throttle use.

### Describe the solution you'd like

For surface controllers (ie Radiomaster MT12) the mixer Slow Up and Slow Down values should have 10ms precision instead of the current 100ms.

### Additional context

Many racer oriented surface controllers implement a servo "speed" setting.  Futaba and Sanwa use a 0-100 percentage style value system while Flysky uses time based values.  This setting can be reasonably emulated using EdgeTX mixer Slow Up/Down settings, but the resolution is not granular enough to dial-in the exact "feel" a driver may be looking to achieve.

Setting on Flysky NB4 for reference:

[image]

Note: Other surface systems use the concept of "away from center" and "return to center" speed settings.  While EdgeTX does not appear to have this concept at the mixer level, I believe that simply setting Up and Down to the same value would be sufficient for most users.

philmoz, 2024-02-06:
Currently the speed up/down values have a range of 0s - 25.5s (stored as an 8 bit unsigned integer).

If we add a setting to allow 10ms precision then you would only get a range of 0s - 2.5s - is this sufficient?

Would the precision setting for speed up/down be for the entire model (affecting all mix lines) or would you want to set it individually for each mix line?

kmf123kmf, 2024-02-06:
In the context of a race vehicle, 2.5s is way more than enough.  With this in mind, I think a setting at the model level would be ideal.  I can't imagine any scenario where someone would need 10ms precision AND super slow movement speed in the same surface model.   

For what it's worth, I was able to hack my way through a POC build for the MT12 and it wasn't too bad.  Just had to watch out because currently the SLOW and DELAY editing seem to be tied to the same code.

kmf123kmf, 2024-02-06:
>I think a setting at the model level would be ideal...

So I've reconsidered this.  The high precision would typically only ever be needed on a steering channel, so the _ideal_ scenario would be a per mixer setting as this affords the highest level of flexibility to the user.  But, if it is easier/more convenient/cleaner/etc. to have single setting per model, this would be perfectly acceptable as well IMO.

philmoz, 2024-02-07:
@kmf123kmf PR #4612 implements this if you want to test it. Precision is set per mix line.

kmf123kmf, 2024-02-07:
Works perfectly on my MT12.  Thanks!

3djc, 2024-02-08:
I would like to add some gtest, but can do as a separete PR

closed 2024-02-12 by pfeerick

e-crins, 2026-01-12:
> Currently the speed up/down values have a range of 0s - 25.5s (stored as an 8 bit unsigned integer).
>
> If we add a setting to allow 10ms precision then you would only get a range of 0s - 2.5s - is [...]

Can this be done the other way around to get a bigger range?
I'm looking for a 0-50s range and precision isn't so much of an issue in my usecase (a long throttle sequence for an indoor model plane) #5736.

PR #4612 (2024-02-12): feat(radio): Allow precision for mix slow up/dn to be set to 10ms or 100ms.
Fixes #4593 

Support the range for slow up/down on a mix line to be 0-2.5s in 10ms increments.