EdgeTX/edgetx #4595: Strango noise from speaker on X10S Express since EdgeTX 2.9.x

steffku, 2024-02-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

There's a light noise coming from the speaker since i use EdgeTX version 2.9 on my X10S Express. The noise starts as soon as i turn on on the transmitter and doesn't stop. I also have a TX16S that doesn't make this sound.

### Expected Behavior

No noise from the speaker.

### Steps To Reproduce

Turn on the transmitter (X10S Express)

### Version

2.9.3

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

pfeerick, 2024-02-02:
Most likey you need to turn on the "Mute if no sound" option in Radio Settings -> Hardware if you have not already done that after upgrading to 2.9. It was a new option added, and prevents the audio being muted like it used to be, to allow for variometer and other time sensitive audio to run without audio mute/unmute delays. Enabling that option restores the previously hard-coded behaviour. 

[image]

steffku, 2024-02-02:
Oh, i wasn't aware of that. Thank you very much for the quick response. I  just wondered that i could hear it on the X10S but not on the TX16S. With the option turned on everythin is fine.

closed 2024-02-02 by steffku