EdgeTX/edgetx #4596: Trim changes with different channel weights.

mrac1, 2024-02-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If you change the weights in the mixer via flight mode or a switch, the trim values change as well. For example, Aileron weight 80/90/100 and trim 10% at 80, then trim changes to 11% at weight 90 and to 12% at weight 100

### Expected Behavior

Trim should not change with different weights.

### Steps To Reproduce

Change different weights with a trimvalue (> 10% is better visible). The same error also occurs with 2.9.2.

### Version

2.9.3

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

3djc, 2024-02-05:
Changing weight in mixer will do that by design, changing weight in input should not

closed 2024-02-05 by pfeerick

mrac1, 2024-02-05:
I see, thanks for the good explanation! :)