EdgeTX/edgetx #4597: Radiomaster Pocket Play Track Special function resets to 1 second repeat after using companion

utahbook, 2024-02-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Special Functions that include "Play Track" do not retain settings of "No repeat" after saving settings back to the radio. Instead, the repeat field goes to 1 second. This happens for all special functions for all models regardless of if that function or model was opened in companion. 

### Expected Behavior

setting for repeat field should stay at "no repeat"

### Steps To Reproduce

write any settings to radio SD card using companion. Observe special functions on radio to see that repeat is now 1 second for play track special functions.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster Pocket

### Anything else?

1/19/24 nightly of 2.10 companion

elecpower, 2024-02-06:
Could be related to the implementation of https://github.com/EdgeTX/edgetx/pull/3601

philmoz, 2024-02-06:
Not reproducible on my TX16S with latest main branch - are you using 2.10 firmware on the radio as well as 2.10 Companion?

pfeerick, 2024-02-07:
Since this is the RM Pocket, you're running a version of 2.10 as it stood around the time the radio was released. But it sounds like are probably still running a version of the firmware on you radio that is older than Dec 3, 2023 with the newer Companion build (You can check that in the SYS -> VERSION page)? 

If so, you need to update your firmware to a newer (i.e. nightly) build. But don't update your Companion though just at the moment... it will not play nicely with the Pocket right now.

utahbook, 2024-02-12:
My apologies, it was indeed the version mismatch between radio and companion. Updated my pocket to the latest nightly and issue is resolved. Thanks, all.

closed 2024-02-12 by utahbook

pfeerick, 2024-02-13:
No problem .. we love problems that fix themselves 🤭 Glad to hear it's working for you now.

Wubbo, 2024-11-18:
I had this problem when I updated my radio from EdgeTX 2.8 to 2.10
The radio is an radiomaster TX16S.
I changed all 45 models in companion by hand.
so now it works again.
I probably used companion 2.8 when this happened, now I use 2.10 
With that same update I also lost a lot of functions, that where visual in the screen but dit not work, I had to delete and than program again.
Also the vario was uncheked with all models that have vario.