EdgeTX/edgetx #4603: Character set bug in opentx-2.9.3 / companion 2.9.3

Freddan-67, 2024-02-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

The bug happens when transferring models back and forth between firmware and companion (windows).
2.9.3 in both firmware and companion
I got a model label called "Sjö" (swedish for sea as in seaplane)
When transferring to companion the label gets duplicated to "SjÃ" and "Sjö". The models gets moved from the label Sjö to "SjÃ".
When transferring back to firmware I got two labels instead of one.

### Expected Behavior

Labels with international character set is kept intact when transferring back and forth between companion and firmware

### Steps To Reproduce

Create a label in the radio with international characters
Assign a model to that label
Transfer models to companion
Transfer back to firmware

### Version

2.9.3

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Windows 11 64bit

### Anything else?

[image]

pfeerick, 2024-02-05:
I suppose I should start with "How did you enter a label of Sjö in the first place?" :laughing: 

And then we can work from there...

Freddan-67, 2024-02-05:
> I suppose I should start with "How did you enter a label of Sjö in the first place?" 😆
> 
> And then we can work from there...

I actually don't remember, but as far as I can tell, the only way I can find the possibility to enter international characters is from label manager in EdgeTX Companion, where I can right click, and then Create Label, and freely name it using windows characters.
Just to see what i got, I tried "åÅäÄöÖ" for all Swedish int. characters, but then that label placed topmost, and the 
model selector in EdgeTX got busted and I couldn't even switch model.
This is how it looks after two more syncs (read models, write models, read modes, write models from CompanionTX)
[image]
I think I need to rename that label, because it keeps getting worse. 

Ver from command line in windows say "Microsoft Windows [Version 10.0.22621.3007]"

Freddan-67, 2024-02-06:
For any developer to test this:
Test 1:
Create a label "Sjö" as in the description
Transfer to the radio
Transfer back to the companion
Make sure "Sjö" didn't duplicate.

Test 2:
Create two models for instance in the companion 
Right click on label manager and create a label. I placed it first among the labels.
Name the label "åÅäÄöÖ"
Transfer to the radio
Now try the model picker in EdgeTX.
Expected behaviour: The model picker can select any model
Actual behaviour: The model picker don't work

philmoz, 2024-02-06:
The radio firmware does not support unicode characters - ASCII only I'm afraid.

Companion should probably be updated to not allow unicode in label / name fields.

Eldenroot, 2024-02-15:
This ugly bug took me 2 hours of my life...had to recreate the complex model from scratch

Eldenroot, 2024-02-15:
We should disable all non-english characters and symbols in model setting...

elecpower, 2024-02-19:
Hi @philmoz. Companion does have a validator for names however it has been turned of or is inconsistent between fields. Can you please provide me with the currently supported characters for B&W and colour radios. Thanks.

3djc, 2024-02-19:
BW allow only basic character set (English), even for localized firmware

philmoz, 2024-02-19:
As far as I know there is no validation on name entry; but only ASCII characters can be entered on the radio.

elecpower, 2024-02-20:
When editing B&W names the radio sim GUI validation only allows space hyphen a-z A-Z 0-9. However as screenshot below from X9D+ they can display an extended set (albeit maybe not all as I have not tested every character possible in colour editor). Should valid B&W character set be extended?

[image]

[image]

Companion validation could be changed to have validation based on the current radio profile ie colour vs B&W.

The next issue is should invalid characters be filtered out when parsing the yaml files?

3djc, 2024-02-20:
A couple more math chars _could_ be added to BW names, not sure they _should_ be tho,

PR #4670 (2024-02-28): fix(cpn): Add radio specific name and label validators
Fixes #4603

Summary of changes:
- change Utf8 to Latin1
- add radio lcd type name and label validators based on firmware permitted characters
- validate names and labels on yaml decode and replace an invalid character with a space. Note: Leading and trailing spaces are removed. 
- use new validators for GUI validation and ensure all name fields are validated
- autolineedit no longer assumes char field is a name. ie now behaves the same as QString field. Up to the dev to assign a validator when required.

Notes
- needs a non-English locale [... cut]
closed 2024-02-28 by pfeerick