EdgeTX/edgetx #4605: serialRead() not working

casfra96, 2024-02-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hi,
I tried to make a widget that receives data from the serial port and print the received data in a label.
The label remains empty.

I 'm using AUX2 (on AUX1 I have the r9maccesmod).
In another script (not enabled at che moment), I use the function serialWrite() on the AUX2 serial port and it works successfully.

The code:

`local received_text = "No data"
local rx_data = ""

local options = {
}

local INDENT = 1 -- left margin in pixels

local function create(zone, options)
  local widget = { zone=zone, options=options }
  return widget
end

local function update(widget, options)
  widget.options = options
end

local function background(widget)
end

function refresh(widget, event, touchState)
  rx_data = serialRead()
    if rx_data ~= "" then
		received_text = rx_data
    end
    lcd.drawText(INDENT, 20, "data: " .. tostring(received_text), LEFT + TEXT_COLOR)

end

return { name="test", options=options, create=create, update=update, refresh=refresh, background=background }`

### Expected Behavior

Received text displayed on text label in the widget.

### Steps To Reproduce

Create the widget.

### Version

2.9.0

### Transmitter

RadioMaster TX16S / TX16SMK2

pfeerick, 2024-02-06:
I can confirm this is still the case in nightly firmware also, so it looks there is an issue still after setSerialBaudrate was fixed in https://github.com/EdgeTX/edgetx/pull/4501. I'm using the following to test both the read and write after having set AUX1 to Lua. serialRead is doing nothing, but serialWrite is working. 

```lua
--- main.lua

local name = "SerialRdWr"
local nameLG = "libGUI"
local libGUI

-- Return GUI library table
function loadGUI()
  if not libGUI then
  -- Loadable code chunk is called immediately and returns libGUI
  	libGUI = loadScript("/WIDGETS/" .. nameLG .. "/libgui.lua")
  end
  
  return libGUI()
end

local function create(zone, options)
  -- Loadable code chunk is called immediately and returns a widget table
  return loadScript("/WIDGETS/" .. name .. "/loadable.lua")(zone, options)
end

local function refresh(widget, event, touchState)
  widget.refresh(event, touchState)
end

local function background(widget)
end

local options = { 
}

local function update(widget, options)
end

return {
  name = name,
  create = create,
  refresh = refresh,
  background = background,
  options = options,
  update = update
}
```

```lua
--- loadable.lua

-- zone and options were passed as arguments to chunk(...)
local zone, options = ...

-- Miscellaneous constants
local HEADER = 40
local WIDTH  = 200
local HEIGHT = 50
local INDENT = 10 -- left margin in pixels

-- serial reading vars
local received_text = "No data"
local rx_data = ""

-- The widget table will be returned to the main script
local widget = { }

-- Load the GUI library by calling the global function declared in the main script.
-- As long as LibGUI is on the SD card, any widget can call loadGUI() because it is global.
local libGUI = loadGUI()

-- Instantiate a new GUI object
local gui = libGUI.newGUI()

-- Make a minimize button from a custom element
local custom = gui.custom({ }, LCD_W - 34, 6, 28, 28)

function custom.draw(focused)
  lcd.drawRectangle(LCD_W - 34, 6, 28, 28, libGUI.colors.primary2)
  lcd.drawFilledRectangle(LCD_W - 30, 19, 20, 3, libGUI.colors.primary2)
  if focused then
    custom.drawFocus()
  end
end

function custom.onEvent(event, touchState)
  if event == EVT_VIRTUAL_ENTER then
    lcd.exitFullScreen()
  end
end

function auxcb()
  serialWrite("Thank you for using EdgeTX!".."\n")
end

local auxWriteButton = gui.button(5, 120, WIDTH, HEIGHT, "Serial Write", auxcb, VCENTER + DBLSIZE + libGUI.colors.primary2)

function gui.fullScreenRefresh()
  -- Draw header
  lcd.drawFilledRectangle(0, 0, LCD_W, 45, COLOR_THEME_SECONDARY1)
  lcd.drawText(INDENT, 23, "Serial port read/write test", VCENTER + DBLSIZE + libGUI.colors.primary2)

  rx_data = serialRead()
  if rx_data ~= "" then
      received_text = rx_data
  end

  lcd.drawText(INDENT, 65, "data: " .. tostring(received_text), LEFT + TEXT_COLOR)

  lcd.drawText(INDENT, 250, "* Ensure either AUX1 or AUX2 is set to LUA!", LEFT + COLOR_THEME_WARNING)
end

-- Draw in widget mode
function libGUI.widgetRefresh()
  lcd.drawRectangle(0, 0, zone.w, zone.h, libGUI.colors.primary3)
  rx_data = serialRead()
  if rx_data ~= "" then
      received_text = rx_data
  end
  lcd.drawText(INDENT, 20, "data: " .. tostring(received_text), LEFT + TEXT_COLOR)

  lcd.drawText(INDENT, 50, "Fullscreen me for more", LEFT + TEXT_COLOR)
end

-- This function is called from the refresh(...) function in the main script
function widget.refresh(event, touchState)
  gui.run(event, touchState)
end

-- Return to the create(...) function in the main script
return widget
```

pfeerick, 2024-02-06:
@raphaelcoeffic DAFAQ? Am I reading this right? No lua serial read if CLI enabled? :exploding_head:  I know you're not to *blame* for this (I do know who _is_ :rofl:) ... but it seems a bit problematic now :zany_face: 

https://github.com/EdgeTX/edgetx/blob/7e706b73d5336bfa086b32913555bf83c5daa837/radio/src/lua/api_general.cpp#L2296-L2298

pfeerick, 2024-02-15:
@casfra96 It would be great if you can try #4644 - here is a 2.9 build if you want to stay there rather than have fun with backuping  up MODEL/RADIO folders due to 2.10 changes... [tx16s-2.9-4644.bin.zip](https://github.com/EdgeTX/edgetx/files/14303458/tx16s-2.9-4644.bin.zip)

PR #4644 (2024-02-26): fix(lua): serialRead is not limited to CLI
Fixes #4605

Summary of changes:
- lua serialRead is not limited to CLI, so shouldn't be disabled when CLI not present (i.e. it is no different to serialWrite). 

It would be interesting to know the justification for this... perhaps it was some attempt at preventing CLI "noise" being read by `serialRead()` - maybe there was a shared buffer at that time?
closed 2024-02-26 by pfeerick