EdgeTX/edgetx #4610: Some radio settings no longer translated

ulfhedlund, 2024-02-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

The Rud, Ele, Thr and Ail text in the Radio settings are no longer translated. It is the same in both the Trainer, Hardware and Calibration tabs.

### Expected Behavior

The text is translated, as in 2.9.3.

### Steps To Reproduce

Set language to anything else than English, open Radio settings, go to the Trainer, Hardware or Calibration tab - and voila!

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

Nightly as per Feb 2, 2024

pfeerick, 2024-02-07:
This is most likley due to #4406 and lupdate needing to be run again. Waiting to see if #4613 fixes it.

ulfhedlund, 2024-02-08:
Yes, lets hope so. However I have run lupdate and updated all SE translations and that didn't work. Hence my post.

ulfhedlund, 2024-02-08:
I'm sure it will, once all JSON-work is finished. I just ran lupdate and translated all new items for SE and as of today it is still the same.

ulfhedlund, 2024-03-09:
Not sure if all planned actions have been taken to fix this issue (the translations) but as far as I can see they are not translated in 2.10 rc1 and the issue remains.

3djc, 2024-03-09:
As you can see, this issue is still open, meaning that no solution has been applied just yet. It will be closed once they are