EdgeTX/edgetx #4614: Gadget showing QR code for latest known GNSS coordinates of (lost) aircraft - for speedy and accurate recovery expedition.

MrMap, 2024-02-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

On the occasions when the flight ended not so well, aircraft downed somewhere not intended, using latest known GNSS coordinates might be the way of recovery. At these occasion a swift transfer of last known coordinates, as presented in Edge TX, can save the day - the sooner the better (successful recovery might be time critical). Though when telemetry is lost, the value characters usually gray out, right when you need it urgently - standing there all stressed up in not so favourable lighting condition. 
So you think you are reading the correct value of GNSS coordinates, but wtf it´s not in the format used in Google maps. Ok, now correct format but you can hardly see what you are typing on your mobile in the intense sunlight... and this sucks.

### Describe the solution you'd like

Generate a QR code depicting the latest known coordinates of the (lost) aircraft. The QR code would be scanned with a smartphone and seconds later you would have the position on a map and then be ready to start navigating there without loosing precious time.
A dedicated gadget would do the presentation in Edge TX.

### Additional context

Once a QR is created, I suggest it remains accessible in the Edge TX gadget even after radio restart - until replace by updated GNSS coordinates. This in order to not inadvertantly loose the relevant QR even i you lost your head and turned the radio off.

pfeerick, 2024-02-07:
Just my 2c... there are already widgets like https://github.com/moschotto/OpenTX_GPS_Telemetry (shows GPS info, has history logging) and https://github.com/kristjanbjarni/opentx-widgets (GPS widget shows GPS info and Google Plus code) ... at the end of the day, just need that Lua widgets/apps can generate/display a QR code and I'm sure the authors of either of these widgets or someone else will implement exactly this.

MrMap, 2024-02-07:
Thank you! I was not aware of these widgets. 
The coordinates QR code would be great evolution on both of them, I think.
As I understand it, QR code generation recently made it´s way into Edge TX master branch. Though as per today not usable within scripts maybe at some point in evolution all pieces will be available to implement my request. 
And yes, maybe the devs you referred to are already on to it...

alufers, 2024-05-28:
https://github.com/alufers/edgetx-gps-qrcode 

I've made a lua script that does exactly that.

MrMap, 2024-05-28:
Wow! I will install and try it.

MrMap, 2024-06-22:
I installed it on RM TX16S MK2, Edge 2.9.4
Copied the LUA into the scripts/tools folder
When running the LUA it just returns "No GPS data yet"

But the handset receives solid 20 satellite lock and coordinates on the standard GPS widget.

Do I need to do anything special to get it to work?

pfeerick, 2024-06-22:
README for that Lua says colorlcd support is a TODO still... i.e. only
talks about tested/supporting B&W screens.

MrMap, 2024-06-22:
> README for that Lua says colorlcd support is a TODO still... i.e. only talks about tested/supporting B&W screens.
> […](#)

AHA!
Well then, sort of explains it.
I hope you eventually have some time to implement on color screens, since it´s very practical.
A few days I had reason after unsceduled landing, I wished I had this widget operational in my TX16S :)

alufers, 2024-06-23:
I tried porting it as a widget for the color LCDs, unfortunately the widget scripts have a limit of how many lua instructions they can execute in one frame (curiously no such limit exists for the B/W radios): https://github.com/EdgeTX/edgetx/blob/22114a67fafb70223cd5ad385d1f162700d9e264/radio/src/lua/lua_widget.cpp#L30

As such, the QR code generation exceeds this limit, because it is expensive to compute. I have tried doing some tricks to adress this (splitting the work into multiple parts executed across multiple frames. Unfortunately I am currently finishing a semester at uni, and on't have the time to continue working on this. Another idea that might be worth is re-writing the QR code generation part in C and contributing it into EdgeTX (Lua realy sucks at it, since it has basically zero native support for byte arrays and bit manipulation).

3djc, 2024-06-23:
A limit exists also on BW, just not at widget level since widgets don't exist in BW

MrMap, 2024-06-23:
> As such, the QR code generation exceeds this limit, because it is expensive to compute. I have tried doing some tricks to adress this (splitting the work into multiple parts executed across multiple [...]

Well, I just wanna recognize you for attempting make it work on color screens even though time is not enough.
And maybe a C coded QR generator would benefit EdgeTX in other ways still not thought of?

pfeerick, 2024-06-23:
It may become possible to use the native LVGL QR code generation soon (if
not already possible) since LVGL widgets have recently been exposed to Lua
scripts ;)