EdgeTX/edgetx #4616: changes to the LED settings are not immediately available

mitchm93, 2024-02-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

if the settings to the LEDs are changed, the new setting is only available after a restart of the radio

### Expected Behavior

if the colors of the LEDs are changed, the change should be available direct after the change - or perhaps after leaving the setting with "RTN"

### Steps To Reproduce

- open the "special functions"
- create a new function (trigger perhaps a trim switch), then select "RGB leds", set it to a color you want
- leave the setting screen

if you trigger the function, nothing happens. The setting is only available after the radio has been restarted

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Anything else?

version: a0ccea1c (2024-02-07)

3djc, 2024-02-07:
I'm (too) aware of it, just haven't had time yet to investigate

philmoz, 2024-02-08:
On color radios and B&W radios with wide screen the code reloads the Lua scripts when a SF/GF is updated that uses a Lua script. I.E. in the onCustomFunctionsFileSelectionMenu function in gui/212x64/model_special_functions.cpp.

This is not done for 128x64 LCD B&W radios in the equivalent file/function. I suspect this is why the change is not being picked up.

Away right now so I can't test this on the MT12.

pfeerick, 2024-02-08:
PL18EV doesn't suffer from this, so it's defintely something in the B&W code, not common code.

pfeerick, 2024-02-08:
This fixes the immediate problem, but a second fix is needed elsewhere... 

```
 void onCustomFunctionsFileSelectionMenu(const char * result)
 {
@@ -76,6 +76,9 @@ void onCustomFunctionsFileSelectionMenu(const char * result)
     // The user choosed a file in the list
     memcpy(cfn->play.name, result, sizeof(cfn->play.name));
     storageDirty(eeFlags);
+    if (func == FUNC_PLAY_SCRIPT || func == FUNC_RGB_LED) {
+      LUA_LOAD_MODEL_SCRIPTS();
+    }
   }
 }
 #endif // SDCARD
 ```

3djc, 2024-02-08:
Not following you, what is this second fix ?

pfeerick, 2024-02-08:
```
        case 5:
            drawCheckBox(MODEL_SPECIAL_FUNC_5TH_COLUMN_ONOFF, y, CFN_ACTIVE(cfn), attr);
            if (active) CFN_ACTIVE(cfn) = checkIncDec(event, CFN_ACTIVE(cfn), 0, 1, eeFlags);
            break;
```

a trigger is needed here also, otherwise again you need to reboot. i.e. it needs to actually be started when you enable the function...

3djc, 2024-02-08:
It is even more than that, if you clear the SF, the former choice keeps running

pfeerick, 2024-02-08:
> @pfeerick <https://github.com/pfeerick> could you try with the default
> script in MT12 ? Does it work when you have on SF line and move from say
> RED to GREEN ? Can't get it right here

Yes, with that change, as long as the trigger condition is true, and SF is
enabled, it changes color immediately once you have selected the file (i.e.
upon pressing enter).

PR #4621 (2024-02-08): fix: load LUA SF after selection
Fixes #4616

Reloads lua scripts on change
closed 2024-02-08 by pfeerick