EdgeTX/edgetx #4618: Allow . (dot, period) in MT12 model names

kmf123kmf, 2024-02-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Many cars and trucks, especially race vehicles, have dots in their names (examples: SC6.4, B74.2).  Currently this character is excluded when editing model names from within the transmitter.  

### Describe the solution you'd like

Add '.' to the list of allowable model name characters when editing within the radio

pfeerick, 2024-02-09:
This is somewhat related to https://github.com/EdgeTX/edgetx/issues/1858

kmf123kmf, 2024-02-09:
> This is somewhat related to #1858

After reading it, implementation of #1858 would be my preferred fix if possible.

PR #5390 (2024-08-08): fix(bw): change char selection order when editing names
Fixes #1858 and fixes #4618 

Change the order of character selection to:
```C
        " abcdefghijklmnopqrstuvwxyz0123456789-,."
```

Use long press to change case.
closed 2024-08-08 by pfeerick