EdgeTX/edgetx #4625: No descriptions of switches and knobs on drop-down lists, Horus X10s with IRSM module

jetfanZK, 2024-02-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hi, 
In my Horus x10s with ISRM ACCESS module, there is no descriptions of switches and knobs on drop-down lists. You can only see arrow symbols or spaces.  The problem occurs in version 2.9.3 and 2.9.2, in earlier versions I have not checked. 
[image]

### Expected Behavior

Descriptions of switches and knobs should be visible in drop-down lists, for example: SA, SB, S1, S2, etc.

### Steps To Reproduce

Visible anywhere there are lists of switches and knobs, for example, in mixers.

### Version

2.9.3

### Transmitter

FrSky X10 / X10S (ACCST)

### Operating System (OS)

Windows

3djc, 2024-02-10:
Have you tried any 2.10 versions (nightly,...) ?

jetfanZK, 2024-02-10:
Not yet, do you think there is a chance that this problem is solved in v 2.10?

pfeerick, 2024-02-11:
I think JC might have been wondering *if* a 2.10 nightly caused it. 

If you go into Radio Settings -> Hardware -> Switches... do the switches have dashes for the names (meaning default names will be used) or are the switch name fields empty (meaning the names have been renamed with spaces), which would explain *why* the names are missing, but not what caused that. And have you just migrated to EdgeTX from OpenTX, or were you using an earlier version of EdgeTX?

jetfanZK, 2024-02-11:
Bingo! Yes, the names are blank! I don't know why this happened. I previously used OpenTX, migrated to EdgeTx in the fall of 2023, then did two system upgrades. Thank you very much for your help!
[image]