EdgeTX/edgetx #4636: Model is not fully imported into transmitter via Companion

Eldenroot, 2024-02-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Check the attached model, try to import it into your transmitter.

Importing wokrs fine, but when you wanna load the model on transmitter, the screen is blank (because no layout for main screen is selected, there should be asome fallback for this scenario).

However the main issue us that "curves", "special functions" etc are not imported. 

They are available in companion when you load the model from Pc, not in the transmitter.

### Expected Behavior

Should be imported, the whole model. If not, some warning should be displayed.

### Steps To Reproduce

Try the attached model, open in companion, then import it to the transmitter and open it.

### Version

2.9.3

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

11 64bit

### Anything else?

[model7.zip](https://github.com/EdgeTX/edgetx/files/14270321/model7.zip)

elecpower, 2024-02-15:
The cause are the Unicode characters in the input names.
Note: It also prevents loading any settings after an incompatible character is encountered.

Test steps:
1. Open the model file in Companion.
2. Edit the input names to remove the Unicode characters.
3. Simulate the model or radio - all settings loaded.

elecpower, 2024-02-15:
Related to #4603