EdgeTX/edgetx #4637: HeadTracker Reset Center

r1000ru, 2024-02-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Any headtracker has drift. To reset the headtracker to zero, you usually press a button on it. This is not convenient, since you need to remove your hand from the remote control. It would be great to reset the headtracker by pressing a remote control button or a toggle switch.

### Describe the solution you'd like

Connection can be made via:
1. Coach port - reset is not possible, since it is one-way.
2. Sbus via serial port (ble or wire) - you can send a magic byte, which the tracker will read and reset.

When using a headtracker, it is common to set one switch to center the camera in one position, or use the headtracker values in another.
As one option, you can transmit the magic byte whenever centering is enabled. This will allow you to combine the position of the camera, head and reset. You could add the switch and its position to send the magic byte in the trainer submenu.

### Describe alternatives you've considered

No alternative

pfeerick, 2024-02-14:
What HeadTracker are you specifically referring to? Is it this one? https://github.com/dlktdr/HeadTracker

As I've not used it myself, but I beleieve it supports SBUS and serial/BT, and has support for "remote" centering, so it may be worth joining the Discord to see how the folks there use it.

arthurson, 2024-02-14:
i use headtracker 2.21 with nano33 ble sense2, and wanna have this function as well

rotorman, 2024-02-14:
See option 3: https://headtracker.gitbook.io/head-tracker-v2.2/getting-started/wiring#option-3-bluetooth-bridge

r1000ru, 2024-02-15:
> See option 3: https://headtracker.gitbook.io/head-tracker-v2.2/getting-started/wiring#option-3-bluetooth-bridge

Do you propose to drill a hole in the control equipment and add a button? Or cut the wires from one of the existing switches and solder them to the bluetooth module?

rotorman, 2024-02-15:
Depends on your radio. E.g. TX16S mkII can have an additional button attached to the back case, where you could route out it's wires below one of the handgrips - or even better - just mount your second Arduino Nano 33 BLE that acts as Bluetooth receiver - internally into the radio. This will result very neat and clean setup. Internally you can then wire up the S.BUS signal to internal USART6/AUX2 header.

See: https://www.radiomasterrc.com/collections/tx16s-mkii-accessories/products/diy-upgrade-button-for-tx16s-mkii-radio

r1000ru, 2024-02-15:
I created the task so that this functionality is available to everyone. I am able to both install the button and modify the software for myself. But it seems to me that the software implementation will be more convenient than the hardware one. And Walksnail also released glasses with a built-in headtracker and bluetooth.  And standardization of the interaction of equipment and glasses - I think it will be very useful.

rotorman, 2024-02-15:
Would you have more info regarding Walksnail Goggles X head tracker? I read the advertisement, but have zero details about it.

arthurson, 2024-02-15:
> Depends on your radio. E.g. TX16S mkII can have an additional button attached to the back case, where you could route out it's wires below one of the handgrips - or even better - just mount your [...]

i now use this method to reset gyro, it works but hard to use as i use two fingers to control gimbal, i would like to use SH to reset if software support it.

r1000ru, 2024-02-15:
Update issue. Proposed a simpler and more functional solution.

3djc, 2024-02-15:
So if I understand correctly, nothing needs doing you can already output channels using SBUS, and you can set any chanel to any specific value of your willing using mixer or SF

arthurson, 2024-02-15:
> So if I understand correctly, nothing needs doing you can already output channels using SBUS, and you can set any chanel to any specific value of your willing using mixer or SF

at the moment when headtracker use sbus to send gyro data to radio like option3, radio can't reset centre by itself, we need to add hardware button to send signal to nana33 ble

3djc, 2024-02-15:
The updated description above suggests 

"send rc channels and headtracker can be configured to reset at a certain value of a certain channel."

arthurson, 2024-02-15:
on trainer menu we have calibration, but thats is not a solution, like you turn left, calibrate,  then turn left, calibrate, repeat serval times, the gyro will not work , only reset button from headtracter can be reliable

r1000ru, 2024-02-15:
> So if I understand correctly, nothing needs doing you can already output channels using SBUS, and you can set any chanel to any specific value of your willing using mixer or SF

Need return channels after mixing to uart->headteacker

pfeerick, 2024-02-15:
> So if I understand correctly, nothing needs doing you can already output channels using SBUS, and you can set any chanel to any specific value of your willing using mixer or SF

Wrong way around JC, headtracker outputs SBUS to handset, which reads it as a trainer input and sends it out via RC link. Headtracker is basically another input. What they are after is a two way link, so that the handset can give instruction to the headtracker. 

I wonder if perhaps a Lua script/widget would do this job? As Lua can send serial commands, and monitor switches... meaning nothing is needed in the firmware at all. 

And we can't do anything for walksnail system until there is documentation as to that sort of API they provide for working with it.

r1000ru, 2024-02-16:
> And we can't do anything for walksnail system until there is documentation as to that sort of API they provide for working with it.

They claim that they cannot start implementing a headtracker because they do not understand which protocol they should support. I suggested that they support two options:
  - emulate FrSky Para trainer (hardware connects to glasses)
  - send SBUS to the WriteNoResponse characteristic of the service 0x0006 of any connected device (connection is carried out with glasses, when searching for available devices, a filter is used for the presence of the service)

rotorman, 2024-02-16:
> They claim that they cannot start implementing a headtracker because they do not understand which protocol they should support. I suggested that they support two options:

Can you please point us to this discussion. Where did you read about this?