EdgeTX/edgetx #4638: SH auto pick in Logical Switches and Special Functions

jmxp69, 2024-02-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

 When creating a logical switch or special function the SH switch is only auto detected with SH down. SH up does not trigger movement in the pick list. In my TX16s I've made SH a toggle and SF is momentary. SF works as expected as do all the other switches. Every position for A-G is auto-picked. Only SH up is not.

### Expected Behavior

In the drop down for selecting a Trigger (SF) or AND (LS) movement of any switch should highlight that switch position in the pick-list.

### Steps To Reproduce

1. Logical Switch - Edit
2. Function a>x
3. v1 Thr Input (or anything else you want)
4. AND Switch pressed
5. SH down shows SH down highlighted in pick-list.
6. SH up does not highlight SH up in pick-list.
7. SA-SG any position always highlights in pick-list.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### OS Version

2024-02-15 02:16:24 (07e0bfa4)

### Anything else?

Looks like I picked a bad week to quit sniffing glue.

pfeerick, 2024-02-15:
Have you made the corresponding changes in Radio Settings -> Hardware to
set SH as  2POS and SF as a toggle switch? SH up not triggering autodetect
sounds like it's still set to toggle, not 2POS.

jmxp69, 2024-02-15:
I did years ago. Somewhere along the line it was lost. Changing it back solves it. Of course now the same question then applies to toggles.

3djc, 2024-02-15:
Toggle are detected only on their active position. If you want to trigger on default position, you can use !

jmxp69, 2024-02-15:
Still inconsistent behavior if every other switch position is highlighted automatically in the pick list when active.

philmoz, 2024-02-15:
> Still inconsistent behavior if every other switch position is highlighted automatically in the pick list when active. I know how to use !thanks.

For a toggle switch I suspect this is by design.

If it detected both states then SH down would be selected when you pull the switch; but then SH up would automatically be selected when you release it - you would then need to tap on SH down or use the scroll wheel to change to it.

Since the active (down) position is probably the only most used it makes send to leave it selected.

3djc, 2024-02-15:
Indeed by design, it IS what makes a toggle different. There might be some confusion tho in authors understanding when I read "I've made SH a toggle and SF is momentary". Momentary switches are toggle and toggle are momentary, they are one thing

jmxp69, 2024-02-15:
> For a toggle switch I suspect this is by design.
>
> If it detected both states then SH down would be selected when you pull the switch; but then SH up would automatically be selected when you release [...]

It feels like a bug to me. Every other switch reacts to a position in the pick list. I feel this one shouldn't be different because it's a toggle.

What is the practical difference between 2pos and toggle? Would anything break in user programming if SH (or in my case SF) were switched to 2pos?

3djc, 2024-02-15:
How will you program it pushed ? Because it will go back to default position when you release your finger, so you have to manually edit the field each time

jmxp69, 2024-02-15:
Right now when a switch field is highlighted and I press the highlight bar in the pick list, the item is chosen and I'm returned to the main screen. So if I toggle/hold/pick, it is selected and I'm returned. That is with SF which is a toggle on my radio and is configured as a toggle. What doesn't auto-highlight is when I release SF. While in the pick list if I release the toggle, the highlight remains on SFdown.

3djc, 2024-02-15:
So it should be no ? SF down is when it is actually activated, isn't it ?

jmxp69, 2024-02-15:
When I pull SF toward me, SFdown highlights automatically in the pick list. 
When I release SF and it moves away from me, SFup does not highlight in the pick list.

3djc, 2024-02-15:
Then set it as a 2 pos, but I'm still unsure why anyone would want to use an SF on default toggle position. The drawback of that is obviously SFlow becomes unselectable that way, since SFup will always replace it

closed 2024-02-15 by jmxp69

pfeerick, 2024-02-15:
> I did years ago. Somewhere along the line it was lost. Changing it back solves it. Of course now the same question then applies to toggles.

As I think has been clearly pointed out by now this is by design, not a bug. Toggle (momentary) switches are only detected by the autoswitch detection on their active state, across *all* radios, not just colorlcd with touch. Thus you don't have to do gymnastics like "hold the switch active and press/touch" in order to select it after it has been autodetected, you just press, release and hit enter/touch it. And if you want the inactive state... well, I know it's so hard to scroll back one entry, or invert it. If there wasn't *some* differences, there would not be any point in having a TOGGLE switch option, would only need 2POS.