EdgeTX/edgetx #4642: ELRS F500 and F1000 only transmit at half rate, when the LUA script is not open

bastian2001, 2024-02-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

F500 and F1000 only transmit at 250Hz/500Hz once the LUA script is closed.

### Expected Behavior

F500 should always transmit at 500Hz and F1000 should always transmit at 1000Hz.

### Steps To Reproduce

1. Open LUA script
2. Set to F1000 if not already done
3. Arm the quad to start blackbox logging
4. Rotate one of the sticks continuously to get a constant stream of new values to look at later
5. While doing that, close the LUA script on the remote (long press RTN)
6. Disarm
7. Look at blackbox logs

Make sure to use F500 or F1000. Everything is fine with D500. Also, this bug only appears when using telemetry. When telemetry is turned off in the LUA script, the full 500 or 1000Hz are used all the time. I recommend turning RC smoothing off to better see the issue. If you want to reproduce this without a battery attached to the drone, follow these steps:
a. betaflight CLI: `set blackbox_mode = ALWAYS`
b. disable bidirectional dshot
c. Check that the only Arming disable flag is `MSP`
d. Go to motors tab, flip the toggle
e. Flip arming switch. This should (visibly in the motors tab) override the motor speeds set in the configurator

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### OS Version

EdgeTX 2.9.2 and 2.9.4

### Anything else?

I first noticed this with [my own FC](https://github.com/bastian2001/Kolibri-FC), where I implemented a packet counter. To make sure that this has nothing to do with the fact that my FC does not yet support all the ELRS features (like FC telemetry to the remote and MSP requests from the RX are completely ignored), I also verified with a normal betaflight FC. Please find below excerpts of the blackbox logs of my FC as well as the logs from the betaflight FC. Roughly at the middle of the picture, the LUA script was closed. It is visible that before this, the packets come twice as often as after closing the script.
[image]
[image]
If you think, this is not actually an EdgeTX error but rather an ELRS bug, please tell me and I will post it here. As far as I understood the CRSF protocol though, the transmitter module (and by extension the RX and FC) are not actually aware of whether the script is open or not. This is why I think it is an EdgeTX bug.

3djc, 2024-02-15:
What is your telemetry rate ? Telemetry packets are taken out of the xxxxHz frames.

Afaik, we are effectively sending at 1000Hz when required by the elrs module

bastian2001, 2024-02-15:
I tried 1:32 and 1:128

bastian2001, 2024-02-15:
I did some more investigation: When I open some other LUA tools on my remote, it also temporarily fixes this bug. E.g. if I open the "Model Locator (by RSSI)", I also get the full 1kHz. When I close the model locator again, the packet rate drops to half rate again (so approx. 500Hz while the setting is F1000). I did these tests at F1000, and 1:128 telemetry rate. When opening no tool, I get half rate, as described above. When I open the Model Locator, ExpressLRS, Multi chan namer, Multi LOLI RX config, MultiConfig or TBS Agent Lite, I get full rate, as long as that specific tool is open. None of the other tools (most of them with a syntax error because I don't have FrSky or that stuff) are not increasing the packet rate to the set speed.

This creates even more questions in my head than it solves. But it makes it more likely that it is in fact an EdgeTX bug unrelated, or only loosely related, to ELRS.

Two more things I forgot to mention before:
- Armed or not (first AUX channel) does not make a difference
- Every 5s, the packet rate jumps to 1kHz, for a brief moment (up to 0.5 seconds) before it goes down to 500Hz again. More precisely: For around 100-500ms the FC gets one packet every ms, and then for the next 5s, there's only one packet every 2ms. Then there's one packet every 1ms for 100-500ms again, and then for the next 5s, there's only one packet every 2ms.

pfeerick, 2024-02-15:
> As far as I understood the CRSF protocol though, the transmitter module
(and by extension the RX and FC) are not actually aware of whether the
script is open or not.

I've seen reference in the ELRS docs that telemetry changes rate when the
Lua is open, so it may be worth asking on the ELRS discord how that is done
and if it relates to this.

3djc, 2024-02-16:
Do you mean when tool is off you stay in tools screen ? This screen is very sd intensive, could you move to a simple main screen instead ?

bastian2001, 2024-02-18:
> asking on the ELRS discord

I asked in the bug-reports channel, but no one really replied :(

> Do you mean when tool is off you stay in tools screen ?

Tools screen or mostly empty home screen doesn't make a difference. Besides, I checked some of the normal rates now as well (neither D nor F). D250 and D500 are fine, yes, but both the F packet rates _and_ the normal packet rates are broken, only D rates work as expected. Extreme case, setting is on 50Hz: While I am in the tool, I am getting the 50Hz. As soon as I leave the tool, it jumps down to 25Hz.

> I'm also puzzled why D500 would make a difference vs F500, since on our side, they are exactly identical

I will put up an issue on ELRS then as well 👍🏻

Some more things: I don't know if this helps in any way, but I checked another ELRS module in the same remote => same issue. Then I switched to a modded FlySky I6X running OpenI6X, a port of a rather old version of OpenTX for that remote. Over there, this was not an issue. Within the LUA script or on the home screen, it was able to sustain full speed. Any ideas?

3djc, 2024-02-19:
Since the radio has no way to know if running F or D, thats very very odd. (both work the exact same way as far as the radio is concerned)
Can you let me know what module (internal ? external ?) you are using and at what speed ?

mha1, 2024-02-19:
Problem solved, culprit is the BF LUA script running in the background. Please close this issue and redirect to BF.

[image]

closed 2024-02-19 by 3djc