EdgeTX/edgetx #4648: feat(telemetry): Edit Sensors: display the percentage (%) equivalent to ratio when scaling a sensor value

frankiearzu, 2024-02-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The original ratio was design to correct for ADC Voltages to display the correct voltage.

Some people who knows how to compute a voltage divider knows what number to type if they know the electrical schematics of the RX voltage input.

For a big majority, they really don't know where to start unless is documented in the RX manual (or somewhere else).  They just start changing the number until they get the voltage they see in a multimeter.

A "ratio" of 0 (display as "-") mean bypass the scaling logic, so it give the raw value coming from the telemetry screen.
A "ratio" of 25.5 will give you the original sensor value (or 100%).   
Could be helpful to display the percentage on the side of the ratio, so if a user needs to scale up, they know to go to a value higher than 100%,

Some people are using "ratio" for sensors who are not voltages. For example, If you have a Current (A) reading, you could compute a new sensor for consumption by adding this current periodically. Then adjust "ratio" +/- until you get a consumption closer to reality at the end of the flight.

### Describe the solution you'd like

Display the equivalent percentage on the side of the "ratio" field.

silicongarage, 2024-02-26:
Hi Frankie,  
I haven't followed the Sensor Scaling saga lately since my model usage is working correctly.  Seems this whole scaling scheme is  getting more and more confusing.   

Is there any underlying code issue that would prevent just having a decimal value used to multiply the sensor input?  
Maybe a value range of 99.99 to 0.01 ?
The default scaling could be the legacy 255 ratio system, with maybe a toggle option to enable a straight forward decimal value multiply?

frankiearzu, 2024-02-27:
@silicongarage   The scaling problems are solved by now (crossing my fingers :-).

Internally as you saw the code is all done by the formula   new_value = (old_value *ratio) / 255.    The original purpose of the code is to adjust Voltages, and there is a formula to compute the ratio depending on RX electronics (resistors used for voltage divider).  There are multiple places documenting what value of "ratio" should be used for different brands of RX (multi-module documentation).

So changing to a totally new formula might affect people who use it, as well as is a deep change in both firmware and companion.   A good trade off is to show the equivalent in percentage for the non electronic inclined users :-)   At least gives you an idea if you are adjusting up or down, and for how much.

silicongarage, 2024-03-01:
Just out of curiosity, what file is the ratio formula for calculating a voltage divider value with ratio being used?

PR #4649 (2024-05-29): feat(radio): modify Sensor Edit screens to show Ratio + Percentage
Fixes #4648
**Works in main as well as 2.9 branch.**
Changes are isolated in the display portion of the UI

Summary of changes:
Change the Telemetry Sensor Edit to show the equivalent percentage.
Works on color and B&W radios.

[image]
[image]

[image]
[image]
closed 2024-05-29 by pfeerick