EdgeTX/edgetx #4650: Model Setup "Throttle Source" not working with CH selections.

frankiearzu, 2024-02-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Similar title to #2520, but not related to Thr Warning.

In 2.9.x,  we are able to select CH3 as a source of the Throttle.  Thr-Cut can be implemented with a SF to override CH3 to -100 with a switch.

This is important is we are computing Timers based on Thr %/Thr start.  If Thr-Cut is on, the Timer should not start even when the throttle stick is moved.

In 2.9.4:
[image]

in 2.10 (nightly as of Feb 16, 2024):
[image]

### Expected Behavior

Should allow CH3  (or any output associated with Thr).

### Steps To Reproduce

1. Create a standard AETR model. CH3 is Throttle output.
2. Create throttle cut:  Special function to override CH3 to -100 when a switch is activated
3. To to Model Setup->Timer1 and set the Mode to "Throttle %"
4. Go to Model Setup->Throttle:    in 2.9.x, you can select CH3 as the source.  In 2.10, you cannot.
 

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-02-16:
You can on BW, so maybe a UI thing. Note that timer also have enabling switches, you could have also you cut there

philmoz, 2024-02-16:
Do you have a mix line defined for CH3 in the model?

mha1, 2024-02-16:
maybe this is related to https://github.com/EdgeTX/edgetx/issues/4339

frankiearzu, 2024-02-16:
Ok, I think I found the problem..  If in your mixers you have a line with no-source, It will not show any CH after that.
For example, add an extra mix to CH2 (Elevator), and don't provide a source (Like Flap->Ele mix)...  
Model Settings Throttle only shows CH1,CH2.

frankiearzu, 2024-02-18:
Probably a corner case, I have a template created that way disabling Flp->Ele mix, and only activate it back if I needed.

I think this part of the code is why it stops looking for used channels after the no-source mix.
[image]