EdgeTX/edgetx #4656: Use ELRS-receiver as transmitter-module via AUX1/AUX2

Steini63, 2024-02-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

ESP8285-based elrs receivers can now also be used as transmitters. All you have to do is flash the appropriate firmware. This is "officially" available via the ExpressLRs configurator (Generic ESP8285 Full-duplex 2.4GHz RX as TX). However, my attempts to operate this on the external module connection of a TX16S have failed.

As I understand it, this is because the UART used for external modules is inverted in the transmitter hardware. The UART of the ESP8285 in the receiver cannot process the inverted serial data. AFAIK neither the ESP8285 nor the processor of the TX16S can be adapted by software.

### Describe the solution you'd like

Is there a chance to use AUX1 or AUX2 for the CSRF signals? 
Then we could use standard ExpressLRF receivers as transmitter modules without extra hardware! The only restriction would be the limitation to 100mW transmission power. But that would be the maximum legal power for Europe anyway and sufficient for most purposes.

pfeerick, 2024-02-17:
This may eventuate in https://github.com/EdgeTX/edgetx/pull/4102/, as this plans to add support for this for use as trainer link, but I don't see why it couldn't be massaged to cover this use case also.

Steini63, 2024-02-19:
Thanks for the reply. I'm sure that would be a great enrichment and a good way to enter the ExpressLRS world with very little effort.

I will keep an eye on the topic and will be ready to test it as soon as there is an implementation.

Frank

3djc, 2024-02-19:
While possible, treating it as a third module has deep implications that need to be taken into account (checklist, EM,....)

Steini63, 2024-02-19:
Would it simplify things to treat it as an alternative external module instead of a third module? The selection for an external module could be extended:
...
- CRSF (via module slot SPort as before)
- CRSF via AUX1
- CRSF via AUX2
...

Frank

SeverFPV, 2024-03-13:
Is it possible to use a trainer port on Taranis QX7 ACCST (3.5 jack or SBUS pin on a JR laptop) to connect internal ELRS? Adjustable power on the RX as TX can be done through the power pins of the lower S.Port (it turns on in the Hardware menu, you need to change "AUTO" to "ON" - but it's inconvenient to do this, besides, the power turns off after a reboot.

pfeerick, 2024-03-14:
> Is it possible to use a trainer port on Taranis QX7 ACCST (3.5 jack or SBUS pin on a JR) to connect internal ELRS

I'm not sure what you are asking for here? trainer port... internal ELRS? 

At the present time, it is the **master/instructor** radio that wants the get the output from the ELRS receiver via SBUS, which you should be able to do with the X7 via the module bay. You then want your slave/student radio to have ELRS Tx so it can connect to that receiver. 

If you want to be able to do ELRS via SBUS for the student radio, no, that's not an option. And, before you ask why not... does ELRS do SBUS in???

pfeerick, 2024-03-14:
Yes, but which way... as a receiver, or as a transmitter? You can presently use it as a receiver, but not as a transmitter.

raphaelcoeffic, 2024-03-30:
Please note that you could do something like that today with small code changes, which you would have to maintain yourself though. You could for instance modify the `_external_ports` array to define only a USART connection to AUX1 or AUX2, and then it would accessible via that port. You will probably need to tweak `_external_module_set_pwr` to achieve what you want.

This is how that structure looks like now:
```c++
static const etx_module_t _external_module = {
  .ports = _external_ports,
  .set_pwr = _external_module_set_pwr,
  .set_bootcmd = nullptr,
  .n_ports = DIM(_external_ports),
};
```

Other than this, as @3djc it's not like we can define an extra module just like that. At least not now.