EdgeTX/edgetx #4659: ABS for surface radio

3djc, 2024-02-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Add support for surface radio (pseudo ?) ABS.

### Describe the solution you'd like

Unsure yet what the form should be (wizard ? )

This is more to gather a better understanding of what ABS is and does for rc cars

### Describe alternatives you've considered

Not doing anything

mawzthefinn, 2024-02-20:
> i played rc ground as well, like drag brake/ boost /turbo are programed by esc but not radio itself.

ABS is however done on the radio level on competing systems.

It's useful for larger aircraft as well, when landing a big scale bird you do not want your brakes locking up. So best implemented as something which can feed an input to the Brake function, be it on the throttle channel or another channel directly controlling brakes.

stokedcrf, 2024-02-20:
I'd also like to stress the importance of this feature as most all other radios have this already.
In most cases, it's a set and forget type of thing (don't really need much adjustability with this one)

fiechr, 2024-02-25:
From the description, a mixer script would probably the best solution, but this has some downsides (less priority, setup more complicated).
https://luadoc.edgetx.org/part_i_-_script_type_overview/mix

A special function on the other hand doesn't allow as many options to enter.

3djc, 2024-02-25:
I have not thought yet on how to actually make it, I thought getting a common understanding of what is needed should be first step. I initially thought there was precise requirements to match something special for the esc, but it looks like a pumping effect is what is actually asked for

pfeerick, 2024-02-25:
The transmitters that I have with ABS option just add a pumping or
"skipping" effect with varying strengths/durations. Which is consistent
with the actual behaviour of anti-skid brakes. Although since it's
transmitter side behaviour obviously there is no integrated feedback loop
to control the pulse duration, frequency, etc.

Wandereradi, 2024-04-09:
> ample

I have ABS on my stock Kyosho KT331 programable tx as a seprate menu option.