EdgeTX/edgetx #4661: Global Function Play Tracks

mfbroberts, 2024-02-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In global functions when you try to select Play Track it says there are no files in the SOUNDS/language directory.
I note several others have found the same issue.

### Expected Behavior

list of the .wav files to come up. It appears that the radio does not look in the correct file location 

### Steps To Reproduce

In Global Functions select Play Track
In Value click file folder

### Version

2.9.3

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

macOS

### OS Version

No it does not, however companion does not obviously put the files in the correct directory

### Anything else?

[image]

mfbroberts, 2024-02-20:
I created the folder SOUNDS/language and put the relevant .wav files in it, but still it said "No files on SD Card!"

frankiearzu, 2024-02-20:
Check that you have files in the "SOUNDS/en" (english version)  folder/directory. Also if you are adding your own .wav files, make sure they that the name is no longer than 6 characters (before the extension).

mfbroberts, 2024-02-20:
Thank you, yes it worked. SOUNDS/en/(.wav files)
Many Many thanks frankiearzu, very much appreciated!