EdgeTX/edgetx #4665: Unable to switch between numbers and variables

chofchop, 2024-02-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

For example, input>TH>Weight, it is not possible to switch between numerical values ​​and variables.

### Expected Behavior

Pressing and holding Enter on this item switches the numerical value <> variable.

### Steps To Reproduce

Even if you try to switch between numbers and variables in an item that can use them, it will not work.

### Version

Other (Please specify below)

### Transmitter

RadioMaster MT12

### Anything else?

I don't know when this phenomenon started occurring, but it has not occurred as of 1b7044a1.

3djc, 2024-02-21:
Just tested current nighties, works normally

mawzthefinn, 2024-02-21:
Works correctly on the shipping MT12 firmware as well in my testing

chofchop, 2024-02-21:
> [1b7044a](https://github.com/EdgeTX/edgetx/commit/1b7044a13465bb84db53e772ec392cdb69480552)

Hello.
I also have no problems with the 1b7044a firmware (January 17th).
I'm converting my display to OLED, so is there any possibility that this may be causing the problem?

chofchop, 2024-02-23:
After deleting radio.yml, normal operation returned.
When I restore the backed up radio.yml, the problem occurs again.
I'll do a little more research to find out what the problem is.

chofchop, 2024-02-23:
It turns out that the direct cause of this problem is in radio.yml.
Probably with firmware after dfaad73,

modelGVDisabled: 0 > No problem
modelGVDisabled: 1 > Can't switch value <> variable

There is no GV Disable checkbox in RADIO SETUP, so it cannot be changed.
If edit radio.yml directly, it will not start with "Missing or bad radio data".

[image]
[image]

3djc, 2024-02-23:
Well makes sense, if GV are disabled, you cannot use those. The question is how did you get it there

chofchop, 2024-02-23:
I actively use GVAR, so I never intentionally disable it.
As mentioned earlier, there is no checkbox for GVDisabesd in RADIO SETUP. Is it in MODEL SETUP?
I started using MT12 two months ago. I've been using the same model data and radio setup ever since.
Isn't it possible that it was set to DVDisabled by default back then?

chofchop, 2024-02-23:
Is it correct that there is no GVDisabled checkbox in RADIO SETUP?
How do I set GVDisabled to 0 in radio.yml?

3djc, 2024-02-23:
It is only available in colorlcd UI, did you go through that at some point ?

chofchop, 2024-02-23:
No, the only radios I own are the X9Lite and MT12.

3djc, 2024-02-23:
When github will have compiled #4671, could you try produced firmware ?

3djc, 2024-02-23:
You can now get it at: https://github.com/EdgeTX/edgetx/actions/runs/8018181460/artifacts/1269425993

chofchop, 2024-02-23:
I compiled #4671 with gitpod (55ea148c).
A checkbox for GROVAL VARIABLES has appeared in RADIO SETUP, but it is still not possible to switch between numbers and variables.
I haven't tested it enough yet, so I'll try it some more.

3djc, 2024-02-23:
Global variable need to be active (either through model or radio), but it works here if active

chofchop, 2024-02-23:
I tried various things and now it works.
Just flashing 55ea148 didn't work. Once I deleted radio.yml and redid the setup, I was able to activate GVAR on each model.
It's good that the problem has been resolved, but it takes a lot of effort to redo RADIO SETUP....

I have a question.
GVAR on/off seems to be saved for each model data, but there is only one GVDisabled in radio.yml. Where is GVAR on/off stored for each model?

3djc, 2024-02-23:
Like all the options, you can choose in your model if you obey radio settings or specific for this model

chofchop, 2024-02-23:
> Like all the options, you can choose in your model if you obey radio settings or specific for this model. 

I don't fully understand it yet, but I found it.

[image]

chofchop, 2024-02-23:
> There as probably no need to redo the radio.yml, you need to understand better how those work

With my current skills, I could only delete it.
I don't want to delete it either, so I'll study it more.

chofchop, 2024-02-23:
Sorry for the late reply, but thank you for resolving the issue.

chofchop, 2024-02-23:
Yes, I intentionally changed it from GLOBAL to OFF.

chofchop, 2024-02-23:
I will ask the remaining questions in the RC groups MT12 thread.

chofchop, 2024-02-25:
I have summarized the history so far.

1.For some reason, GVDisabled in my radio.yml was set to 1. At this point, the firmware was ignoring GVDisabled, so there was no problem.
2.Since dfaad73, the firmware no longer ignores GVDisabled in radio.yml. At this point I can no longer use GVAR.
3.GVAR checkbox was added to RADIO SETUP in #4671, but this operates on GVDisabled in modelxx.yml, so I still cannot use GVAR.
4.I manually edited GVDisabled in radio.yml to 0. I am now finally able to use GVAR.

In my opinion, the GVAR checkbox in RADIO SETUP should operate on radio.yml. If GVAR need to turn on/off for each model data, I think the checkbox should provide in MODEL SETUP.

3djc, 2024-02-25:
After discussion, we have decided to prevent disabling GVAR on 128 for now. PR has been updated accordingly

chofchop, 2024-02-25:
That's good.

PR #4671 (2024-02-26): fix: ignore invalid GVAR feature flag on BW128 UI
Looks like GVAR option UI was missed out of #3484 for BW128 UI.

It doesn't map to a screen in 128 UI, but can block feature if imported from another radio 

Fixes #4665
closed 2024-02-26 by pfeerick