EdgeTX/edgetx #4667: Audio & Haptic settings not saved

oomwat, 2024-02-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I edit the Radio Settings and update the Audio and Haptic configuration setting them to Off, this correctly turns them off, however when I restart the transmitter they are reset back to their default values.

### Expected Behavior

I would expect the audio settings to be saved between restarts of the transmitter.

### Steps To Reproduce

1. Plug the transmitter into a computer using USB
2. Enter Radio Settings on the transmitter (Stock literadio 3 pro)
3. Edit the Audio and Haptic settings to Quiet
4. Relax now that the incessant warnings have finally stopped [(](https://github.com/EdgeTX/edgetx/issues/4666)
5. Switch the transmitter off
6. Repeat Step 1

### Version

2.9.3

### Transmitter

Other (Please specify below)

### Anything else?

My spidey senses suggest it may be a similar class of issue to this one: https://github.com/EdgeTX/edgetx/issues/2158

oomwat, 2024-02-22:
After further testing I was able to store the changes after reboot by using the EdgeTX Companion app ... so it's unlikely to be 'off' colliding with 'undefined' after all.

pfeerick, 2025-02-14:
Closing as unable to reproduce with 2.10.6. There was some issues early on with the LR3 and some of the storage settings, so it was most likely fixed there.

closed 2025-02-14 by pfeerick