EdgeTX/edgetx #4669: [NV14] horizontal lines

nightflasher, 2024-02-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

There are three horizontal lines in the bottom of the screen. They only disappear (fade away) in Bootloader mode, but appear as soon the NV14 is started regularly. 

### Expected Behavior

No horizontal lines when not needed!? 

### Steps To Reproduce

Turn on NV14 with edgeTX 2.9.x (maybe earlier version had this too?) 

### Version

2.9.3

### Transmitter

Flysky NV14

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04 LTS

### Anything else?

Find the display code from Bootloader and check against regular display driving code. 
[image]
[image]

I've tried different firmware versions, but only down to 2.9.0 beta (or alpha?)

skazi1982, 2024-02-24:
It's probably a defect of the screen or the screen tape, I have something like this in tx16s, banggood refunded me some of the money for it.

nightflasher, 2024-02-25:
Yours is defective, you can see shifted pixels, mine just have the lines and are not affecting the Bootloader screen. A hardware defect can't be there in one screen/mode and is gone in another.

pfeerick, 2024-02-25:
Is it only present at the main screen, or when you are in the various
settings screens also? If it is only at the main screen, it could be the
that the background image needs replacing/updating (it was changed a while
ago IIRC due to alpha transparency/blending issues).

nightflasher, 2024-02-26:
Made a short video https://vimeo.com/916706610
Didn't record other screens, but it's present in all screens in normal operation mode, disappear on bootloader.