EdgeTX/edgetx #4672: Trainer: Master/CPPM module not working on TX16s

mha1, 2024-02-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Can't establish Master/CPPM module trainer connection

### Expected Behavior

Feeding CPPM signals to the module bay's HB pin as trainer input should work

### Steps To Reproduce

- use a receiver capable of outputting CPPM to the module Bay's HB pin
- enable trainer mode Master/CPPM module
- define SF to activate trainer mode
- observe no trainer connection be established

### Version

2.9.3

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

Looking a bit deeper: not a single interrupt for TIM4_CH1 is generated (TIM4_IRQHandler in trainer_driver.cpp) while a valid 8 channel CPPM signal (20ms period, 0.5ms sync) is fed to the HB pin.

My CPPM signal ought to work, right?

[image]

mha1, 2024-02-26:
I'll close this issue as "can't be fixed on TX16s" due to a RM design f'up with heartbeat signal shaping and with saying to RM: "You can do better than that". I also leave this for documentation should the question "I can select Master/CPPM Module trainer mode on my TX16s but it won't work" be asked again: https://github.com/EdgeTX/edgetx/pull/477#issuecomment-884311379

closed 2024-02-26 by mha1

pfeerick, 2024-02-26:
> asked again: #477 (comment)

I believe this is a shared T16/TX16 issue (not just RM) and was fixed in
the TX16 MK2, so RM already have done better. 🤣

gagarinlg, 2024-02-26:
Well, almost. It still does not work with some signal sources, like the T14 3.5mm output

mha1, 2024-02-26:
> I believe this is a shared T16/TX16 issue (not just RM) and was fixed in the TX16 MK2, so RM already have done better. 🤣

I don't think so. Copying a bad design without realizing it's bad doesn't win you awards. And they didn't fix the design. They just changed the diode to a Shottky type which still has enough voltage drop to cause problems. The "done better" solution would've been to include a proper circuit in a PCB redesign. At least they could've taken the minimal risk and replaced the diode with a zero Ohm resistor.  But hey, it's CPPM, a relict of the old days and thanks to you being very kind SBUS is working soon. 👌

pfeerick, 2024-02-26:
Touche, maybe I should have said "improved", not "fixed" 😅. Maybe something for the MK3?... gotta keep them wanting more 🤪🤣