EdgeTX/edgetx #4680: What is Reg.ID? The edgetx wiki does not explain it.

AExY1j1, 2024-02-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Some model files have this item, while others do not.
[image]
[image]

### Describe the solution you'd like

Explain what the project does.

AExY1j1, 2024-02-28:
The version is v2.9.4

pfeerick, 2024-02-28:
Reg(istration) ID is pretty much specific to Frsky, for their ACCESS RF system. It's used so that receivers can be "registered" to multiple transmitters (when the same registration id is used). It's marketed as being there so that if you are racing, you have multiple transmitters set up, and if one transmitter goes flat or is faulty, you just pick up another and carry on.

pfeerick, 2024-02-28:
Still only Frsky radios for the most part - since it is dependent on PXX2 ;)

closed 2024-02-28 by pfeerick