EdgeTX/edgetx #4681: There is a bug in 'Build Instructions under Ubuntu 20.04'

nostromo123, 2024-02-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

There seems to be a bug in

https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04

I am here under Debian 12, but using a fresh xubuntu 20.04.4 in a VM for building.

1. Setting up the build environment for EdgeTX

looks ok!

2. Building EdgeTX firmware for the radio

also ok!

3. Building Companion, Simulator and radio simulator libraries

but both stop with ERROR!

Here the terminal protocol of 'Setting up the build environment for EdgeTX'
[BE 29.2.2024.txt](https://github.com/EdgeTX/edgetx/files/14445407/BE.29.2.2024.txt)

and Building EdgeTX firmware and companion
[FW and Companion.txt](https://github.com/EdgeTX/edgetx/files/14445420/FW.and.Companion.txt)

My first test was with an old xubuntu 20.04 (2022), I still hold on my system, so today I created a fresh xubuntu 20.04.4 (for cross checking) and have the same ERRORs in companion and Sim.

That is the ERROR part in building companion

```
/home/vbox/edgetx/edgetx_main/companion/src/storage/labeled.cpp: In member function ‘virtual bool LabelsStorageFormat::loadBin(RadioData&)’:
/home/vbox/edgetx/edgetx_main/companion/src/storage/labeled.cpp:98:57: error: ‘SkipEmptyParts’ is not a member of ‘Qt’
   98 |     QStringList parts = line.split(QRegExp("\\s+"), Qt::SkipEmptyParts);
      |                                                         ^~~~~~~~~~~~~~
[ 18%] Building CXX object companion/src/storage/CMakeFiles/storage.dir/yaml.cpp.o
[ 18%] Building CXX object companion/src/storage/CMakeFiles/storage.dir/crc.cpp.o
[ 18%] Building CXX object companion/src/storage/CMakeFiles/storage.dir/minizinterface.cpp.o
[ 18%] Building CXX object companion/src/storage/CMakeFiles/storage.dir/moc_appdata.cpp.o
make[7]: *** [companion/src/storage/CMakeFiles/storage.dir/build.make:197: companion/src/storage/CMakeFiles/storage.dir/labeled.cpp.o] Error 1
make[7]: *** Waiting for unfinished jobs....
make[6]: *** [CMakeFiles/Makefile2:2799: companion/src/storage/CMakeFiles/storage.dir/all] Error 2
make[5]: *** [CMakeFiles/Makefile2:1032: companion/src/CMakeFiles/companion.dir/rule] Error 2
make[4]: *** [Makefile:317: companion] Error 2
make[3]: *** [CMakeFiles/companion.dir/build.make:57: CMakeFiles/companion] Error 2
make[2]: *** [CMakeFiles/Makefile2:306: CMakeFiles/companion.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:313: CMakeFiles/companion.dir/rule] Error 2
make: *** [Makefile:222: companion] Error 2
vbox@vbox:~/edgetx/edgetx_main/build-output$ 
```

Please have a look.

br KH

My aim is to test, if it still is possible to bypass this https://github.com/EdgeTX/edgetx/issues/1435 by creating my own companion under an actual Debian release. Was lucky with Edge 2.6 and 2.7, but therefore I have to have a working ubuntu building environment ....

### Expected Behavior

 Build Instructions under Ubuntu 20.04 working! ;-)

### Steps To Reproduce

See attachments

### Version

2.9.3

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Linux

### OS Version

xubuntu 20.04.4 (Debian 12 bookworm)

### Anything else?

Had an issue with building before ... https://github.com/EdgeTX/edgetx/issues/1327

With that xubuntu I did my tests yesterday. 

[Building FW CP 28.2.2024.txt](https://github.com/EdgeTX/edgetx/files/14445675/Building.FW.CP.28.2.2024.txt)
[Building SIM 28.2.2024.txt](https://github.com/EdgeTX/edgetx/files/14445683/Building.SIM.28.2.2024.txt)

ERROR companion
```

/home/kalle/edgetx/edgetx_main/companion/src/storage/labeled.cpp: In member function ‘virtual bool LabelsStorageFormat::loadBin(RadioData&)’:
/home/kalle/edgetx/edgetx_main/companion/src/storage/labeled.cpp:98:57: error: ‘SkipEmptyParts’ is not a member of ‘Qt’
   98 |     QStringList parts = line.split(QRegExp("\\s+"), Qt::SkipEmptyParts);
      |                                                         ^~~~~~~~~~~~~~
[ 18%] Building CXX object companion/src/storage/CMakeFiles/storage.dir/crc.cpp.o
[ 18%] Building CXX object companion/src/storage/CMakeFiles/storage.dir/minizinterface.cpp.o
[ 18%] Building CXX object companion/src/storage/CMakeFiles/storage.dir/moc_appdata.cpp.o
make[7]: *** [companion/src/storage/CMakeFiles/storage.dir/build.make:197: companion/src/storage/CMakeFiles/storage.dir/labeled.cpp.o] Fehler 1
make[7]: *** Auf noch nicht beendete Prozesse wird gewartet …
make[6]: *** [CMakeFiles/Makefile2:2799: companion/src/storage/CMakeFiles/storage.dir/all] Fehler 2
make[5]: *** [CMakeFiles/Makefile2:1032: companion/src/CMakeFiles/companion.dir/rule] Fehler 2
make[4]: *** [Makefile:317: companion] Fehler 2
make[3]: *** [CMakeFiles/companion.dir/build.make:57: CMakeFiles/companion] Fehler 2
make[2]: *** [CMakeFiles/Makefile2:306: CMakeFiles/companion.dir/all] Fehler 2
make[1]: *** [CMakeFiles/Makefile2:313: CMakeFiles/companion.dir/rule] Fehler 2
make: *** [Makefile:222: companion] Fehler 2
kalle@kalle-VirtualBox:~/edgetx/edgetx_main/build-output$
```

ERROR SIM
```
/home/kalle/edgetx/edgetx_main/companion/src/storage/labeled.cpp: In member function ‘virtual bool LabelsStorageFormat::loadBin(RadioData&)’:
/home/kalle/edgetx/edgetx_main/companion/src/storage/labeled.cpp:98:57: error: ‘SkipEmptyParts’ is not a member of ‘Qt’
   98 |     QStringList parts = line.split(QRegExp("\\s+"), Qt::SkipEmptyParts);
      |                                                         ^~~~~~~~~~~~~~
make[7]: *** [companion/src/storage/CMakeFiles/storage.dir/build.make:197: companion/src/storage/CMakeFiles/storage.dir/labeled.cpp.o] Fehler 1
make[6]: *** [CMakeFiles/Makefile2:2799: companion/src/storage/CMakeFiles/storage.dir/all] Fehler 2
make[5]: *** [CMakeFiles/Makefile2:1059: companion/src/CMakeFiles/simulator.dir/rule] Fehler 2
make[4]: *** [Makefile:330: simulator] Fehler 2
make[3]: *** [CMakeFiles/simulator.dir/build.make:57: CMakeFiles/simulator] Fehler 2
make[2]: *** [CMakeFiles/Makefile2:279: CMakeFiles/simulator.dir/all] Fehler 2
make[1]: *** [CMakeFiles/Makefile2:286: CMakeFiles/simulator.dir/rule] Fehler 2
make: *** [Makefile:209: simulator] Fehler 2
kalle@kalle-VirtualBox:~/edgetx/edgetx_main/build-output$ 
```

nostromo123, 2024-02-29:
I am using

`git clone --recursive -b main https://github.com/EdgeTX/edgetx.git edgetx_main`

So this is not 2.9 ??

I'll do a quick check with

 `git clone --recursive -b main https://github.com/EdgeTX/edgetx.git edgetx_2.9`

and come back.

Sorry, old guy here ;-)

Here is the test. Again ERROR with companion.

[test with 2.9.txt](https://github.com/EdgeTX/edgetx/files/14446442/test.with.2.9.txt)

3djc, 2024-02-29:
20.04 doesn't support the required QT version. You can use qt utilities to install the right version, but it doesn't support scripting as you need to create an account with qt to get that specific version.

Alternarive is to use 22.04 which supports the required qt

nostromo123, 2024-02-29:
OK. Found a fault on my side. Sorry!

Did the 2.9 test with

vbox@vbox:~/edgetx$ git clone --recursive -b **main** https://github.com/EdgeTX/edgetx.git edgetx_2.9

A new test this afternoon.

nostromo123, 2024-02-29:
> 20.04 doesn't support the required QT version. You can use qt utilities to install the right version, but it doesn't support scripting as you need to create an account with qt to get that specific version.
> 
> Alternative is to use 22.04 which supports the required qt

but that would mean, the instruction should be revised. It states **Ubuntu 20.04!**

nostromo123, 2024-02-29:
1. I'll repeat the test with 

git clone --recursive -b 2.9.4 https://github.com/EdgeTX/edgetx.git edgetx_2.9

(found a note in my documentation, last time I did this 
git clone --recursive -b v2.7.1 https://github.com/EdgeTX/edgetx.git edgetx_v271)

2. Create a building environment with xubuntu 22.04 and see, how it 'behaves' there ...

3djc, 2024-02-29:
No, branch is 2.9, NOT 2.9.4

nostromo123, 2024-02-29:
OK, give it a try.

About testing on a 22.04, I get that here. (Have a fresh 22.04 VM here)
```

vbox@vbox:~$ wget https://raw.githubusercontent.com/EdgeTX/edgetx/main/tools/setup_buildenv_ubuntu20.04.sh
--2024-02-29 14:54:42--  https://raw.githubusercontent.com/EdgeTX/edgetx/main/tools/setup_buildenv_ubuntu20.04.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5247 (5,1K) [text/plain]
Saving to: ‘setup_buildenv_ubuntu20.04.sh’

setup_buildenv_ubun 100%[===================>]   5,12K  --.-KB/s    in 0s      

2024-02-29 14:54:42 (47,5 MB/s) - ‘setup_buildenv_ubuntu20.04.sh’ saved [5247/5247]

vbox@vbox:~$ ./setup_buildenv_ubuntu20.04.sh
ERROR: Not running on Ubuntu 20.04!
Terminating the script now.
vbox@vbox:~$ 

```

Think, for working under 22.04, the whole setup has to be revised ...

3djc, 2024-02-29:
already available : https://github.com/EdgeTX/edgetx/blob/main/tools/setup_buildenv_ubuntu22.04.sh

nostromo123, 2024-02-29:
I'll test later. Thanx!

nostromo123, 2024-02-29:
Back to 20.04 with that command line here

`vbox@vbox:~/edgetx$ git clone --recursive -b 2.9 https://github.com/EdgeTX/edgetx.git edgetx_2.9`

Building FW ok!

but now it is stuck here
```
[100%] Built target firmware
vbox@vbox:~/edgetx/edgetx_2.9/build-output$ cd arm-none-eabi
vbox@vbox:~/edgetx/edgetx_2.9/build-output/arm-none-eabi$ mv firmware.bin edgetx_2.9.4_tx16s_lua-ppmus-mode2_debug.bin
vbox@vbox:~/edgetx/edgetx_2.9/build-output/arm-none-eabi$ cd ..
vbox@vbox:~/edgetx/edgetx_2.9/build-output$ make -j`nproc` companion
[100%] Built target native-configure
Scanning dependencies of target companion
[100%] Built target companion
vbox@vbox:~/edgetx/edgetx_2.9/build-output$ make -j`nproc` companion29
make: *** No rule to make target 'companion29'.  Stop.
vbox@vbox:~/edgetx/edgetx_2.9/build-output$ make -j`nproc` companion2.9
make: *** No rule to make target 'companion2.9'.  Stop.
vbox@vbox:~/edgetx/edgetx_2.9/build-output$ 

 
```

I'll test now 22.04 with the 'ubuntu22.04.sh'

nostromo123, 2024-02-29:
22.04 with that script looks far better ;-)

[image]

Think, you should call that 20.04 Linux building instruction obsolete.

My task now, converting that building environment to Debian 12 bookworm ....

Think, we can close that now.  Thank you for giving a hand!!

br KH :-)

nostromo123, 2024-02-29:
But there is one additional question I have.

Building  'main' is working, but when I try it with loading this 

`vbox@vbox:~/edgetx$ git clone --recursive -b 2.9 https://github.com/EdgeTX/edgetx.git edgetx_2.9`

I am stuck at that point after building the firmware and trying to build companion.

```
[ 98%] Building CXX object radio/src/CMakeFiles/firmware.dir/targets/horus/keys_driver.cpp.obj
[100%] Building CXX object radio/src/CMakeFiles/firmware.dir/targets/horus/diskio.cpp.obj
[100%] Linking CXX executable ../../firmware.elf
[100%] Built target firmware
[100%] Built target firmware

vbox@vbox:~/edgetx/edgetx_2.9/build-output$ cd arm-none-eabi
mv firmware.bin edgetx_2.9.4_x10express_lua-ppmus-mode3_release.bin

vbox@vbox:~/edgetx/edgetx_2.9/build-output/arm-none-eabi$ cd ..

vbox@vbox:~/edgetx/edgetx_2.9/build-output$ make -j`nproc` companion

[100%] Built target native-configure
[100%] Built target companion <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
```
The instruction reads

> 
> Building Companion, Simulator and radio simulator libraries
> To build Companion binary, issue in the same terminal as opened previously:
> make -j`nproc` companion

and that step works fine with 'main'.

What I am doing wrong here??

jtaylor2, 2024-02-29:
You need to be in the native directory when you issue the make companion command.
cd to native

pfeerick, 2024-02-29:
> You need to be in the native directory when you issue the make companion command. cd to native

No, you only need to be in the `native` folder to run `simu` / `companion` if you want the correct relative paths... for building either firmware or companion you **must** be in the main/base build folder (i.e. `~/edgetx/edgetx_2.9/build-output`)

> But there is one additional question I have.
> 
> I am stuck at that point after building the firmware and trying to build companion.

Try running `make configure` or `make native-configure` and then try ``make -j`nproc` companion`` again... perhaps some companion dependencies are missing or screwed up. If it still doesn't work, empty the `build-output` directory (i.e. `rm -R *` while in that directory) and try with a new `cmake ... ` and `make ...`

note that in the following:
```
cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=3 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Release ../
```

- `-DGVARS=YES` is redundant as it is enabled by default
- `-DPPM_UNIT=US` is no longer valid as it is a runtime option now - (hence the cmake warning about it being unused)
- `-DDEFAULT_MODE=3` since I mentioned runtime (configurable on the radio) options, just want to point out this is a runtime option also, at the bottom of the radio setup page - but it does let you set it manually
- `-DLUA_MIXER=YES` is redundant as it is enabled by default

so that prior line can be reduced to the following with no change in behaviour of the resulting firmware. 
```
cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=3 -DCMAKE_BUILD_TYPE=Release ../
```

In fact, (`-DCMAKE_BUILD_TYPE=Release` _could_ be dropped also as I think it is only used to remove debug symbols from the simulator build`, but it don't hurt to have it there anyway - it has zero effect on a normal firmware build, but may help reduce the size of a companion/simulator build).

nostromo123, 2024-03-01:
Thank you for your note about the radio conf. command.
Don't have a TX16 here and that were my notes 2 years ago for building ...
```

git clone --recursive -b v2.7.1 https://github.com/EdgeTX/edgetx.git edgetx_v271
cd edgetx_v271 && mkdir build-output && cd build-output
cmake -LAH ../ > ~/edgetx_v271-cmake-options.txt
cmake -DPCB=X10 -DPCBREV=EXPRESS -DHELI=NO -DLUA=YES ../
mv firmware.bin edgetx_v271_X10_Express_lua-noheli.bin
make -j`nproc` companion27
make -j`nproc` libsimulator && make -j`nproc` simulator27
```
Had to add a `make configure` and I think I remember it worked fine with 2.8 ....

But back to the problem with building companion and sim.

Started the xubuntu 22.04 'fresh', created the 'building environment' and had the 1st. run with building 'main'. Then the 2nd. run with 2.9. But please have a look yourself ... (I omitted the first part 'building the firmware' and added blank lines at commands, so it might make it easier to read the text)

[New File.txt](https://github.com/EdgeTX/edgetx/files/14462401/New.File.txt)

br KH

nostromo123, 2024-03-01:
One idea.

I am using Xubuntu 22.04.4 with the setup script, 3djc send. Could it be, that this combination is not building companion and simulator with an 'old' version like 2.9.4, but no problems with 2.11.0?