EdgeTX/edgetx #4692: LUA: special function repeat parameter missing for model.setCustomFunction

ParkerEde, 2024-03-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In the current main, no value can be set for "Repeat" via lua model.setCustomFunction. I think this has happened since there is now a separate activation switch

### Expected Behavior

repeat like  !1x, 1x, 1s, 2s and so on should be setable for model.setCustomFunction

### Steps To Reproduce

try to set repeat like  !1x, 1x, 1s, 2s and so onwith  model.setCustomFunction in a lua script

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

current main

philmoz, 2024-03-04:
Repeat and active used to share the same data field in the custom function. Now they are separate.

There was never a 'repeat' entry in the table so you would have had to set it using the 'active' property in the Lua script.

I can add a 'repeat' entry to the table - you will need to update any scripts though.

ParkerEde, 2024-03-05:
Yes, exactly. Before, for example, activate=255 => !x1 
Since this is now split up, it would be super helpful if we could introduce a new attribute, e.g. "repeat".

pfeerick, 2024-03-05:
Does the link PR do what you want?

ParkerEde, 2024-03-05:
Basically, it has been implemented as I imagined. Unfortunately, SET does not yet work. I have described it in more detail in the PR

PR #4693 (2024-03-05): fix(lua): add 'repetition' property to getCustomFunction and setCustomFunction tables
Fixes #4692
closed 2024-03-05 by pfeerick