EdgeTX/edgetx #4695: MT12 delta<=x (or delta<x) function operation MIA in logic switch config

TodWulff, 2024-03-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

There is no option to select delta<=x (or delta<x) function when configuring a logic switch

### Expected Behavior

A user should be able to select to compare for a negative delta and trigger a logic switch if true

### Steps To Reproduce

1 - enter logical switch config
2 - scroll through the function options
3 - note absence of delta<=x (or delta<x) function option 

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Anything else?

MT12 ELRS Transmitter reports:

FW: edgetx-mt12
VERS: 2.10.0-nightly
GIT#: d6db7465
DATE: 2023-12-22 02:22:44

Note: Tx shipped from RM with this build thereon.  I haven't yet updated, pending the release of a stable build.

philmoz, 2024-03-06:
If you want to detect a negative delta change just enter a negative value for V2.

E.G. 
Δ>x where V1 = THR and V2 = -10 will detect decreases in throttle movement.
Δ>x where V1 = THR and V2 = 10 will detect increases in throttle movement.

pfeerick, 2024-03-06:
I also think this has been brought up before, and wasn't going to change as it doesn't make sense... delta is change, so you either have 0 (no change) or the amount of change, whether it be in the pos or neg direction. If you really wanted the opposite, I guess you could use the 'not' state of the logical switch (i.e. `!L01` instead of `L01`)... thereby negating the need for anything more than the current comparison?

TodWulff, 2024-03-06:
Ya know, I have been heads down in the radio UI a ton these last couple of days and, frankly, I am embarrassed that I didn't even think to try a -X value as an operand.  Further, I broke out one of my Air TX16S' and that too doesn't have a Δ<x function option.  So, I'll close this and then grab a napkin to wipe the egg from my farce.  Thanks.  Sorry for not seeing the forest due to all the trees in my way.  Take care.

closed 2024-03-06 by TodWulff

pfeerick, 2024-03-06:
lol... no napkin required... just sweat and perseverance... :hand_over_mouth: