EdgeTX/edgetx #4696: Trim configs on DM0 (maybe other DM_ too.?.) config page allow for selection of non-existent T6-T8 (presumably)

TodWulff, 2024-03-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On DM0 configuration page, the trim switch config selection scrolls off the screen when selecting what would be T6-T8 and then reappears with either left or right scrolls of an appropriate amount.

Note: Doing this, and editing something non-existent, appears to possibly be related to the frustrating audio being forced off.  It 'feels' like a register or ram value is getting clobbered and borking the audio (and possibly other things.?.), but that is just cursory observation/supposition, no hard evidence yet - too afraid to get back into a no-audio situation...

### Expected Behavior

When on T5 config field and scrolling to the right one click, the cursor should move to the Fade In value field.
When on the Fade In value field, scrolling to the left once click, should have the cursor move to the T5 config field.

### Steps To Reproduce

1 - Enter into DM0 config page
2 - Scroll right to select T5 config field
3 - Scroll right one, two and three more times - cursor is off into the ethers presumably trying to config T6-T8
4 - Scroll right one more time and cursor magically appears on the Fade In value field.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Anything else?

MT12 ELRS Transmitter reports:

FW: edgetx-mt12
VERS: 2.10.0-nightly
GIT#: https://github.com/EdgeTX/edgetx/commit/d6db74659fe9db8b0f460e7b3a0020985cab81f1
DATE: 2023-12-22 02:22:44

Note: Tx shipped from RM with this build thereon. I haven't yet updated, pending the release of a stable build.

TodWulff, 2024-03-06:
Sorry guys, this is actually a duplicate of issue 4615 (forgot to check closed issues).

closed 2024-03-06 by TodWulff

pfeerick, 2024-03-06:
All good... means it *should* be fixed in a nightly firmware ;)