EdgeTX/edgetx #4700: v2.10.0-RC1 Issue Tracker

pfeerick, 2024-03-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

a

### Expected Behavior

b

### Steps To Reproduce

c

### Version

Other (Please specify below)

### Transmitter

Other (Please specify below)

pagrey, 2024-03-10:
Keys don't seem to work in the simulator for 9x navigation radios like the tlite. I've tried 2.10 and pfeerick/basic-simulators with the same result.

pfeerick, 2024-03-11:
`pfeerick/basic-simulators` should have no impact on this, it is just adding those four simulators + enabling Companion support for those radios. With 2.10-0-rc1 I can confirm the T-Lite simulator visually responds to clicks of the buttons (or keyboard shortcut presses) but simulator is not responding, and is common to all `NAVIGATION_TYPE 9x` simulators  :confounded: 

x7, x9d, xlite and horus navigation types all appear unaffected... :open_mouth:

djabmal, 2024-03-11:
hello, its not an issue i just suggest an improvement if its possible. We use a lot of different curves in our models and sometiment we build curves that we no longer use . Would it be possible to see the curves that are really in use in one model? It could be similar to the logical switches tool were you can visualize the logical switch in use.
thanks
mal

patman13, 2024-03-14:
@pfeerick Why do you not support MT12 in the companion?

pfeerick, 2024-03-14:
Because the changes needed to both companion and simulator to support a completely different interface style of radio are not trivial.

closed 2024-03-28 by pfeerick

pfeerick, 2024-03-28:
RC2 will be tracked in #4817