EdgeTX/edgetx #4702: Glider wizard corrupts model program

fhpage, 2024-03-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Current version of Edgetx with TX16S.
Created a model with glider wizard
Power off radio and then power back on, the model file is corrupt, void of most of the settings.
Happens for every model file created with the wizard.

### Expected Behavior

DON'T corrupt the file!

### Steps To Reproduce

Create a model with glider wizard
power off radio
power on radio
observe that model file is void of a name, and most entries.

### Version

2.9.4

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-03-08:
Just followed those steps:
- flash TX16S with Buddy on 2.9.4
- format sdcard, use Buddy to make a TX16S card
- run radio, do calibration
- create new mode
- use wizard, select glider
- complete the glider wizard:
[image]

- turn off radio
- turn on radio : model is absolutely fine, nothing to report

Model created by wizard and copied after reboot is attached
[model2.zip](https://github.com/EdgeTX/edgetx/files/14538262/model2.zip)

fhpage, 2024-03-08:
Well, I replicated it multiple times with two different TX16S.
I can send details and samples.

3djc, 2024-03-08:
Yes please, because with the procedure above, it does work fine. Please ensure you are using 2.9.4 sd content, because older might create issues

frankiearzu, 2024-03-08:
Just test it with 2.9.4. works as expected, and created the glider correctly. I can see inputs and mixers.
Are you changing the option of "No, I need to change something" to "Yes, is all well..."..  without confirming the last step, it will not create the model.

fhpage, 2024-03-16:
After 5 days model files are OK. Appears was not a bug, not hardware.

closed 2024-03-16 by fhpage

pfeerick, 2024-03-17:
Thanks for the update. Hopefully it was just some corruption in the wizard SD card files you have previously installed and it's gone now.