EdgeTX/edgetx #4713: Support for external XJT Module on Jumper T20

dadantsou, 2024-03-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There is no support for XJT when selecting protocol for external module on Jumper T20

### Describe the solution you'd like

I wish an implementation that can activate XJT protocols for external module

pfeerick, 2024-03-10:
XJT or XJT Lite?

dadantsou, 2024-03-10:
XJT

pfeerick, 2024-03-10:
Out of curiosity, how are you attaching the JR Micro XJT to the T20? Custom adapter? As I thought there was only a Nano form factor module bay available for it?

3djc, 2024-03-10:
Wouldn't a nano multi make a much better job without the utter ugliness of a JR module on T20 ?

dadantsou, 2024-03-10:
I don't want to buy another XJT module, I want just use it for my old Frsky receiver.
If possible, I will design later a slimmer case for it

pfeerick, 2024-03-10:
Hm... interesting. 

Custom firmware with `MODULE_SIZE_STD` instead of `MODULE_SIZE_SML` maybe?

3djc, 2024-03-10:
yes; was going to suggest he forks and make custom firmware with that change, understanding that it closes the door to other modules

dadantsou, 2024-03-10:
OK, thanks guys

pfeerick, 2024-03-10:
Do you have a link to that adapter btw? Looks handy for some Frankenstein testing purposes :zany_face:

dadantsou, 2024-03-10:
I've found it on Aliexpress.
[https://www.aliexpress.com/item/1005004743012876.html?spm=a2g0o.cart.0.0.4b9238dagJDQSu&mp=1](url)

pfeerick, 2024-03-11:
Thanks for that... I was hoping it was going to be something could 3d print myself, but still might be handy. 

@3djc What do you think about an option in the radio settings that makes it possible for all RF module types the radio can actually work with selectable? Otherwise, I think we can safely leave this as a "too much of an edge case" scenario, with a custom firmware build being most appropriate.