EdgeTX/edgetx #4724: HID Device Descriptor reports wDescriptorLength 83 instead of 56

ThomasKuehne, 2024-03-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

querying the USB interface via `lsusb -d 1209:4f54`  reports an inaccurate wDescriptorLength of 83 bytes for the HID Report Decriptor:

```HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      83
```

The actual length for the classic Joystick HID descriptor is 56 and dynamic for the advanced Joystick.

### Expected Behavior

device reports accurate lenth for the HID Report Decriptor

### Steps To Reproduce

connect device as a joystick to a computer and run lsusb and usbhid-dump

### Version

Nightly (Please give date/commit below)

### Transmitter

Jumper T-Lite

pfeerick, 2024-03-14:
That's strange... it's correct on the X9D+2019  and TLite with 2.10.0-rc1 and nightly for me... is there something more to this than it just being in classic mode?

```
Device Descriptor:
  iManufacturer           1 OpenTX
  iProduct                2 Jumper TLite Joystick
```
```
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      56
```

```
Device Descriptor:
  iManufacturer           1 OpenTX
  iProduct                2 FrSky Taranis Joystick
```
```
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      56

```

ThomasKuehne, 2024-03-15:
The bug report is buggy. However the underlying issue is real.

works as expected:
1. When a PC asks for the USB configuration descriptor (LIBUSB_DT_CONFIG) the radio returns `USBD_HID_CfgDesc`.
2. `USBD_HID_CfgDesc` contains an embedded HID descriptor.
3. When a PC asks for the HID descriptor (LIBUSB_DT_HID) the radio returns `USBD_HID_Desc`.
4. When a PC asks for the HID report descriptor (LIBUSB_DT_REPORT) the radio returns `usbReportDesc()`.
5. The report length specified in `USBD_HID_CfgDesc` (2) matches the length of the actually returned descriptor (4).

Unexpectedly:
6. The length specified in the standalone HID descriptor (3) doesn't match the length specified in the embedded HID descriptor.

 I forgot that I'm using a heavily modified lsusb. Replication with standard lsusb isn't possible.

Standalone [USB-HID-descriptor-lint](https://github.com/ThomasKuehne/USB-HID-descriptor-lint) to verify sanity of the HID descriptor:

PR #4726 (2024-04-05): fix(radio): usb joystick descriptor, unusable joystick axis 
general code review of radio/src/targets/common/arm/stm32/usbd_hid_joystick.c

Fixes #4724
Fixes #4725

Summary of changes:
- code de-duplication
- LOBYTE / HIBYTE coding ensuring that more complex joysticks can implemented -not yet critical for main branch but already encounted in other branches
closed 2024-04-05 by pfeerick