EdgeTX/edgetx #4729: Telemetry: Negative floats are incorrectly decoded

blopa1961, 2024-03-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The Archer SR8 receiver reports a range of -180.00 to 180 degrees for Roll and Pitch (0 is level).
EdgeTX telemetry decodes left roll as 655.99 to 477.00 instead of negative values (Archer SR10+ also affected).
These last values probably correspond to signed -0.01 to -179.99 using a limited number of bits where the sign is the 10th binary digit of the integer part (in two's complement?). The same happens with Pitch. It's a formatting (sprintf?) problem for negative float numbers.

The problem is that it's not possible to convert those values to something useful by changing the ratio and offset in the telemetry sensor.

I tested both SR receivers in Ethos and correctly got negative values for left roll (and back pitch).

### Expected Behavior

I expect LEFT roll values to be negatives in the range of -0.01 to -180.00

### Steps To Reproduce

Bind transmitter to an Archer receiver with stabilization like SR8 or SR10+
Go to telemetry, discover new sensors and roll the receiver right and left. Notice negative values are wrong.

### Version

2.9.4

### Transmitter

FrSky X9 Lite / Lite S

pfeerick, 2024-03-13:
> I tested both SR receivers in Ethos and correctly got negative values for left roll (and back pitch).

Thanks for this confirmation... I have noticed that the IMU values from my SR10+ didn't seem right, but didn't have any point of reference for comparison to know where the problem stemmed from.

frankiearzu, 2024-03-13:
I can take a look, but don't have that receiver.
Can you tell me the ID and SubId of the sensors.   If is using Sport Telemetry, should be 0730 01 or something like that.

Seems that the values are 16 bits number, but the value 655.99 is bigger than 16bits (0x1003F). 477.00 is 0xBA54 if treated signed would be -178.36.

Looking at the code, is only treating it as an unsigned number, where is converting from a uint_16 to a uint_32.

[image]

Maybe by just type casting it to int_16 will do the trick:

[image]

frankiearzu, 2024-03-13:
@blopa1961 can you just confirm the sensor IDs, and I think this change will work. I can compile the code with the change for you to test since I don't have that sensor.

pfeerick, 2024-03-13:
On SR10+ Pitch/Roll indeed is 0730 26 for both

[image]
[image]

pfeerick, 2024-03-13:
I've only got a minute to test, but it looks like your change fixes it... roll was giving me up to 180 and then going negative.

frankiearzu, 2024-03-13:
Do you want to do the PR or want me to do it??

pfeerick, 2024-03-13:
You fixed it, it's your pr ;)

blopa1961, 2024-03-13:
> @blopa1961 can you just confirm the sensor IDs, and I think this change will work. I can compile the code with the change for you to test since I don't have that sensor.

Sorry for the delay, I'm in another time zone.
Roll and Pitch have the same ID: 730-26

AccX: 700-26, AccY: 710-26, AccZ: 720-26
But AccX, AccY and AccZ correctly report negative floats for g. 
Example: AccZ= -0.98g
Both AccX and AccY are 0.00 when the receiver is not shaken

frankiearzu, 2024-03-13:
once the tests finish running, there should be an artifact created with all the different firmware.
https://github.com/EdgeTX/edgetx/pull/4733/checks
But this is for the main branch (2.11) that is still in development and/or could be merge into 2.10.
If you needed urgently, will need to merge/build an X9 version of 2.9.4 or 2.10 with this change.

blopa1961, 2024-03-13:
> If you needed urgently, will need to merge/build an X9 version of 2.9.4 or 2.10 with this change.

Nah, I can live without an artificial horizon LOL. Will eagerly wait for 2.10
Thank you for fixing it so fast!

blopa1961, 2024-03-14:
@frankiearzu this has been very interesting, thank you for your patience, explanations and the fix!

after looking at the frsky_sport.cpp code you posted I now understand that:
1. my report of this bug should have stated a range starting in 655.35 (-0.01) instead of 655.99 and it's not a formatting problem (it's typecast)
2. telemetry is received as 32 bit streams
3. the stream can contain up to two 16 bit values (Roll and Pitch have the same ID of 0730-26 in FrSky's stabilized receivers)
4. interpreting the values depends on the dataID of the packet with which you identify:
4.1 if it's a 32 bit signed or unsigned value
4.2 if it contains 2 signed or unsigned values (example: Roll and Pitch have exactly the same ID and they are signed 16 bit integers in the SR8)
4.3 if it's a float and how many of the digits are decimals, thus automatically changing the "precision" value to move the decimal point to the left (and this is why you are so interested in IDs and expected values of unknown sensors)
Why is this not a table indexed by dataID and containing number of bits, number of packed values, if signed or unsigned and number of decimals (0= integer)? This would let us define unknown sensors on the fly...

Thus:
- In this bug, Roll and Pitch were correctly identified as floats with 2 decimals (which updated precision to "0.00") but the value was taken as unsigned when it was signed
- the Matek G474's Roll and Pitch are in a range of -1800 to 1800 because the values were correctly decoded as signed but precision was not updated to 1 decimal because the ID was unknown: Roll 5230-0 (int w/1 decimal), Pitch: 5240-0 (int w/1 decimal), VFAS: 210-0 (uint w/2 decimals), Hdg: 840-0 (uint w/2 decimals).
- iNav expects Roll and Pich to be in the range -1800 to 1800 because it was developed with precision's default of "0" when it should have been "0.0"

blopa1961, 2024-03-14:
@frankiearzu This is the sensor ID data I collected for the hardware I have; I hope it's useful. If you want the worksheet let me know.

[image]

PR #4733 (2024-03-25): fix(telemetry): frsky angles should be be signed (allow negative)
Fixes #4729

Summary of changes:
Bug report that Roll/Pitch should be from -180 to +180, so needs to be a signed 16 bits number.
**Change Angles to be processed as Signed int16**
closed 2024-03-25 by pfeerick