EdgeTX/edgetx #4731: Frozen UI when USB-CDC Telemetry Mirror without connection

r1000ru, 2024-03-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If you assign USB-CDC as a Telemetry Mirror, connect the wire and select USB-CDC in the menu, then the interface freezes before connecting to USB Serial from the host side. If you disconnect from the USB Serial, the interface also freezes. If you pull out the cord without disconnecting, the interface comes to life.

### Expected Behavior

UI (and may be somethink else?) not freezes.

### Steps To Reproduce

1. In hardware setup set USB-CDC is Telemetry Mirror.
2. Turn on transmitter and receiver, wait connect.
3. Plug usb cable from PC to transmitter and select USB-CDC.
4. Check, that UI is frozen.
5. Open USB Serial port in terminal
6. You will see a stream of bytes of telemetry
7. Check, that UI is live
8. Close USB Serial
9. Check, that UI is frozen.
10. Unplug usb cable
11. Check, that UI is frozen.
12. Plug usb cable and reopen the USB Serial port in the terminal
13. Check, that UI is live
14. Unplug cable (do not close the connection in terminal)
15. Check, that UI is live

### Version

Other (Please specify below)

### Transmitter

RadioMaster Pocket

### Operating System (OS)

Windows

### Anything else?

EdgeTX 2.10-RC1

philmoz, 2024-03-13:
Reproducible on TX16S and MT12 (have not tested others).
In my case both TX16S and MT12 completely lock up and require the battery to be pulled to recover.

raphaelcoeffic, 2024-03-15:
@philmoz and we agree this is not an issue caused by any UI code, right?

philmoz, 2024-03-15:
> @philmoz and we agree this is not an issue caused by any UI code, right?

It happens on both color and B&W radios so I don't think it is UI related.

PR #4822 (2024-03-29): fix(usb): do not wait for send state
Fixes #4731
Fixes #4740 

Summary of changes:
- removed wait for send state from `usbSerialPutc()`.
- use `usbPlugged()` instead of `cdcConnected` as the later is updated from the UI task which is blocked while in passthrough mode.
commit 06c09579 (2024-03-29): fix(usb): do not wait for send state

Fixes #4731
commit 429356a7 (2024-03-30): fix(usb): do not wait for send state

Fixes #4731
commit 417c37ce (2024-03-30): fix(usb): do not wait for send state

Fixes #4731
closed 2024-03-29 by pfeerick