EdgeTX/edgetx #4736: Throttle Warning Button missing on  Model setup page > Throttle

Airfiddler, 2024-03-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Unable to turn off the throttle warning nag, unless I go to Companion and deselect
the throttle warning feature. Users should be able to select this feature (on or off )on the transmitter.
Several guys at our field are also frustrated with this missing option.

### Describe the solution you'd like

Install throttle warning option button on model setup page- throttle page.
I saw this feature earlier, but it seems to be missing now. 2.9.4
Thanks for all the great work on Edge TX. :-}  Don

philmoz, 2024-03-14:
I'm guessing this is a B&W radio where the Pre-start Checks section is now collapsed by default (to declutter the UI).
Press the down arrow button to the right of 'Pre-start Checks' to expand it.

[image]

[image]

Airfiddler, 2024-03-14:
Tx16s

pfeerick, 2024-03-14:
I edited my original reply, so am partially repeating it here as you may not have got it via email:

Are you sure you're going to the right page? Throttle warning is part of the preflight/prestart checks, and so the option to the throttle warning on or off is on that page, not on the throttle page. It's visible for me on TX16S running 2.9.4 as "Throttle state" switch on the Preflight checks page.

for colorLCD
https://manual.edgetx.org/v/v2.9/edgetx-user-manual/user-manual-for-color-screen-radios/model-settings/model-setup/preflight-checks

Airfiddler, 2024-03-14:
Ahh yes I see it now. It's working. But why "Throttle warning" on Companion and "Throttle state on the TX16s" ? somewhat confusing. I like the term warning, as there are no options on the "Throttle State" button. Just On/Off. The most logical place to look for this function would be model setup, then the "throttle"  screen. 
Maybe the terms could be changed to match, and an additional function button could be put on the "Throttle" page. 
Just my 2 cents worth, from a user not a programmer.    Thanks again  Don

pfeerick, 2024-03-14:
Not sure why it is "Throttle warning" in Companion. That does need changing so they are aligned - and yes, it is probably better worded as "warning", not "state". But as you will have noticed, Companion and radio firmware don't necessarily group everything together quite the same way. 

As far as moving it, I can't say I agree. It would have been grouped under Preflight Checks back in the OpenTX days (as it has always been where it is - it hasn't been moved) because it is indeed a preflight check, rather than a throttle "option". It also seems a bit redundant adding it in a second place. Additionally, the option underneath would need to be moved/duplicated also, as the custom position setting is related to the the throttle warning check also. However, the feedback is appreciated, and will be taken into consideration.

closed 2024-03-14 by pfeerick

Airfiddler, 2024-03-14:
Now I see "Throttle State" brings up the "Custom Position" option. On the
TX16s.

So it works slightly different than "throttle warning "on Companion.
Maybe an additional "throttle state" button on Companion and a throttle
warning on the transmitter would clarify things for dummies like me.