EdgeTX/edgetx #4740: Radio freezes when updating internal ELRS via USB-VCP

pfeerick, 2024-03-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With the Zorro and T14, updating internal ELRS via USB-VCP locks the radio up hard, requiring removal of the battery. The ELRS pass through update is successful, the radio UI simply stops updating from the moment the update starts. I've not tried this on colorlcd yet to see if this is present there also, implying a common lower level issue. 

### Expected Behavior

That the radio not require a hard power off when using passthrough updating. 

### Steps To Reproduce

1. Connect USB, select USB-VCP
2. Perform passthrough update using ExpressLRS Configurator
3. Witness successful update, unresponsive radio

### Version

Other (Please specify below)

### Transmitter

RadioMaster Zorro

### Anything else?

v2.10.0-rc1

philmoz, 2024-03-14:
Possibly related to #4731

pfeerick, 2024-03-14:
Yes, sounds very similar.

KondratokGK, 2024-03-22:
Same issue, but ELRS is not flashed

commit de2e2f9c (2024-03-29): fix(usb): use `usbPlugged()` to detect disconnect

`cdcConnected` is updated from the UI task and cannot be used here.

Fixes #4740
PR #4822 (2024-03-29): fix(usb): do not wait for send state
Fixes #4731
Fixes #4740 

Summary of changes:
- removed wait for send state from `usbSerialPutc()`.
- use `usbPlugged()` instead of `cdcConnected` as the later is updated from the UI task which is blocked while in passthrough mode.
commit 2d4c1acb (2024-03-29): fix(usb): use `usbPlugged()` to detect disconnect

`cdcConnected` is updated from the UI task and cannot be used here.

Fixes #4740
commit 64410eff (2024-03-30): fix(usb): use `usbPlugged()` to detect disconnect

`cdcConnected` is updated from the UI task and cannot be used here.

Fixes #4740
closed 2024-03-29 by pfeerick