EdgeTX/edgetx #4752: Companion: missing elements in mixer lines

CopterFail, 2024-03-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

The companion does not show direct stick inputs in mixer lines imported from the T20. But this works for stick inputs defined in the input section.

### Expected Behavior

The mixer lines in companion were identical to the local BW menu of the T20

### Steps To Reproduce

1. define a mixer line in the T20 using direct stick input
2. connect the T20 as storage
3. start companion and read the according model
4. use the editor to view the mixer line
5. the defined input is empty

### Version

Other (Please specify below)

### Transmitter

Jumper T20/T20S

### Operating System (OS)

Linux

### OS Version

linux mint 21.3

### Anything else?

Version: 2.10.0RC1

elecpower, 2024-03-22:
Did you use a T20 radio profile? If so, how, since there is no support in 2.10 RC 1. You would get error messages when trying to use a T20 profile.

[image]

Ignoring the errors and powering on will produce undetermined results.

CopterFail, 2024-03-22:
Yes, this is exactly what happens.

elecpower, 2024-03-22:
Support for T20 expected to be in 2.10 RC2.

closed 2024-03-22 by elecpower