EdgeTX/edgetx #4754: Bluetooth Multiple/Universal AT Command Support

packetkiller, 2024-03-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Wider range of Bluetooth device support is sought.

### Describe the solution you'd like

Specifically for telemetry, support for any Bluetooth device, no matter the AT command set, by abstracting the AT commands via something like an external reference table or file (one file per AT command set). The only requirement would be support for the minimal required features needed for telemetry streaming (slave mode, baudrate, flow control, etc.).

### Describe alternatives you've considered

Attempting to reverse engineer the few specific bluetooth devices and associated AT command set supported by EdgeTX.

### Additional context

The concept is to allow additional bluetooth device support without having to rewrite or hard code AT command set syntax inside of EdgeTX. It also allows less guess work on the end user about how the bluetooth device needs to be configured to work with EdgeTX.

frankiearzu, 2024-03-25:
I was playing with an AT-09 (cone of HM-10) and there are variations in firmware between the vendors. 
Telemetry Mirror usually work on any since the phone/device connects to it, and you pre-confugured the settings externally before connecting the device.  
Trainer is a bit more elaborated.  I was able to change all the AT-commands to the HM-10 equivalent, but there were weird/different behavior between two AT-09s.  One responded immediately to the AT commands.. the other was out for 2-3s on AT commands who change configuration (bound rate, role) and i think the firmware was resetting the device.  At the end implemented a try mechanism to get it to work, but only for Trainer Master (as originator of the connection)...   as a slave, there was a problem... the AT-09 did not send a "Connected" string when a new connection happened.. it just start to send data. MH-10 is suppose to have a setting for this announcements, but is not implemented in the AT-09.

Besides the details, I think the challenge is that we can't get a consistent firmware, even for the same hardware.  Practically we will need to adapt the code and behavior for device+firmware that  will make it un-maintainable. 

At the end, I ended up using the Bluetooth device implemented by the HeadTracker team.. for the TX16 MkII they uses a ESP32 board with custom firmware that behaves like the original FrSky Bluetooth device.  Probably an opensource approach like that could be a way forward. Standarize the hardware, and upgrade the firmware via the TX  (similar approach to the multi-module),

rotorman, 2024-03-26:
For ESP32 based Bluetooth module, have a look at: https://btwifimod.gitbook.io/