EdgeTX/edgetx #4759: 2.10rc1 conversion issue from 2.9.4

jmxp69, 2024-03-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I flashed 2.10rc1 on top of 2.9.4 and in my Avanti Jet configuration, I lost RS and LS entries in the mixer, logical switches, and special functions. In the mixer I saw RS (gear) replaced with --  And LS (flaps replaced with -- in multiple locations throughout the configuration.

### Expected Behavior

After flash, specified controls (sliders in my case) remain throughout the configuration.

### Steps To Reproduce

Use the 2.9.4 yaml, on 2.9.4 firmware and verify LS/RS entries in mixer/LS/SF. 
Flash 2.10RC1
Install Companion 2.10
In Companion read models settings. Delete a few models but do not make any changes to Avanti configuration.
Write models/settings to radio.
Note loss of LS/RS in mixer/LS/SF.

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

macOS

### OS Version

2.10RC1 and 2.10 Companion for Mac.

### Anything else?

[Archive.zip](https://github.com/EdgeTX/edgetx/files/14628942/Archive.zip)

gagarinlg, 2024-03-17:
Oh, it was with Companion.
@elecpower is this already fixed in a PR?

jmxp69, 2024-03-17:
I wasn't smart enough to review the model settings on the radio before working in Companion. I probably should do that.

I found another issue after making field edits. Previously there were logical switches set to override the gear channel 12 with 100 or -100. After the migration, Ch12 became Ch16 in the configuration. I just corrected that since I opened the issue.

elecpower, 2024-03-17:
> Oh, it was with Companion. @elecpower is this already fixed in a PR?

There is a fix for reading binary files.

jmxp69, 2024-03-17:
This was a 2.9.4 radio updated to 2.10 and with 2.10 Companion on Mac.

elecpower, 2024-03-17:
> I wasn't smart enough to review the model settings on the radio before working in Companion. I probably should do that.
That would be helpful if you could check the radio before the settings are read into Companion.

> I found another issue after making field edits. Previously there were logical switches set to override the gear channel 12 with 100 or -100. After the migration, Ch12 became Ch16 in the configuration. I just corrected that since I opened the issue.

It would be very helpful if you can upload the unbroken radio.yml and Avanti Jet modelnn.yml files ie do not use Companion.

jmxp69, 2024-03-17:
Both yml files are in the archive.zip of the original post. 2.9.4 and 2.10 post flash.

philmoz, 2024-03-17:
I tried loading the 2.9.4 version file on my TX16S and it looked ok.

So probably just some remaining Companion gremlins.

elecpower, 2024-03-17:
> Both yml files are in the archive.zip of the original post. 2.9.4 and 2.10 post flash.

Oops my bad missed them

elecpower, 2024-03-17:
Companion 2.9.4 reads model 2.9.4 LS & RS
Companion 2.10 RC1 reads model 2.9.4 LS & RS
Companion 2.10 RC1 mis-reads model 2.10.RC1 LS & RS

The 2.9.4 model SFs display CH16 in Companion 2.9.4 and the yaml file has zero based index 15 which is 16 displayed

jmxp69, 2024-03-17:
I restored my radio to pre-2.10rc1.
I verified (took screenshots) of the configuration in Companion 2.9 and on Radio.
I flashed 2.10rc1
I verified configuration on Radio and RS/LS were there in the mixes and ls/sf. 
I read models/settings from Companion 2.10 and verified the configuration again. Nothing missing.
I wrote models settings from Companion 2.10 to radio and the radio looks ok.

So I haven't been able to replicate whatever it was that happened--but something did happen because I had to re-program every use of RS and LS in the mix/ls/sf. 

The Ch16 override was present in 2.9 too, so not sure where that crept in, but that should be Ch12 override (gear). So let's scratch that off the list for this issue.

elecpower, 2024-03-18:
Thank you for checking, it is much appreciated. I believe the issue is Companion reading the radio generated 2.10 yaml. The radio can handle the Companion variant but not the reverse. Leave it with me to fix.

elecpower, 2024-03-19:
> So I haven't been able to replicate whatever it was that happened--but something did happen because I had to re-program every use of RS and LS in the mix/ls/sf.

Okay so after going in circles for hours testing a myriad of combinations between Companion, sims and versions I believe I have the combination. Also, my statement 'Companion 2.10 RC1 mis-reads model 2.10.RC1 LS & RS' was due to opening your model file directly ie without a radio.yml file.

Steps:
1. Flash the radio to v2.10 RC1
2. Allow the radio to start normally but do not select and load any other models
3. Read the models and settings into Companion 2.10 RC1
4. Open models other than the one active on the radio post the firmware update.
5. Some settings will likely be broken.
6. Write models and settings to radio and the corruption will follow.

Cause: if you do not load a model on the radio after flashing the firmware, the non-active model files do not have their version updated and settings converted. However the radio.yml and the active model files will have their version updated and any conversions performed. When Companion reads the models and settings it currently only checks the radio.yml version to decide whether any conversions are necessary to radio and individual model files.

When Companion writes the models and settings to the radio it updates the version to the Companion version across ALL files and writes the data (valid or not) to the files.

Thus the reason for the warnings that come with every release about backups and checking all models and settings.

jmxp69, 2024-03-19:
Wow

I'm glad to know at least you were able to uncover a scenario leading to breakage.

Regarding the user perspective: I figured after updating to 2.10, I'd take it as an opportunity to go in and clean out some old models/labels. I like Companion for that type of maintenance. It sounds like the process you documented accurately outlines how it unfolded.

MrMap, 2024-03-21:
I had a similar experience updating TX16S from 2.9.4
Also on another TX16S that I updated from OTX 2.3.11 and then imported the model file saved from within OTX. Big mess like described above.
Companion on Win11.

MrMap, 2024-03-21:
I hesitate at what remedy I should try to use
Rollback Companion and firmware to 2.9.4
or wait for a new 2.10 Release Candidate with a fix?

elecpower, 2024-03-21:
2.10 RC2 will be out soon with a fix provided testing goes ok. However it will not fix corrupted files so you will need to restore your backed up pre 2.10 files and reload into the fixed version of Companion.

PR #4768 (2024-03-24): fixcpn): use model semver over radio semver for decoding model sources and switches
Fixes #4759 and also display message for each model file where the version is ahead of Companion.
closed 2024-03-24 by pfeerick