EdgeTX/edgetx #4761: When using Flysky digital Hall universal joint, jumper T16 calibration is invalid and cannot be used

beach97, 2024-03-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When using Flysky digital Hall universal joint, jumper T16 calibration is invalid and cannot be used

### Expected Behavior

The next version will solve this problem

### Steps To Reproduce

EdgeTX "Centurion" v2.10.0-RC1
When using Flysky digital Hall universal joint, jumper T16 calibration is invalid and cannot be used

### Version

Other (Please specify below)

### Transmitter

Jumper T16

### OS Version

EdgeTX "Centurion" v2.10.0-RC1

pfeerick, 2024-03-18:
FYI, this is not specific to the Jumper T16 but rather the flysky "digital" hall gimbals generally, and is a known issue specifically referenced in the release notes. Whether this can be remediated is unclear at this point, as it will need to be tied to upgrading from 2.9 to 2.10, and should be a one-off occurrence.

richardclli, 2024-03-21:
This change is essential to expose the full potential of the high resolution Flysky gimbal.  And a recalibration is required because the full range is used instead of pre-scaled one (only has 2048 res).

closed 2024-04-08 by pfeerick