EdgeTX/edgetx #4762: 'Edit GVAR' screen should display FM numbers as well as names

RC-SOAR, 2024-03-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Users of my glider templates are having issues with the GVAR menus. These are due to an inconsistency between the main grid and the edit menu:

The main grid displays flight modes by their number only, no mention of FM names:
[image]

However, in the Edit menu, the FM names are displayed, but _not the flight mode numbers_.
[image]

This is confusing for first time users, especially as there's nothing to separate the FM lines from the header block.

### Describe the solution you'd like

In the GV edit screen, all flight modes should prefixed by the number FMn: for example:

FM0:Thermal
FM1:---
FM2:---
FM3:Cruise

I've only looked at this for the colour LCD screens, but I think the same should apply to the mono screen radios. If there's a choice of displaying the FM name or the number, IMO the number is more important.

pfeerick, 2024-03-18:
It says GV4=FM0 in the page header... does that correctly update when you
change flight modes? I don't disagree though, index should be shown for
consistency as well as name (since now have space to show it).

RC-SOAR, 2024-03-18:
Yes, the top line updates correctly when the active flight mode is changed.

>  I don't disagree though, index should be shown for
> consistency as well as name (since now have space to show it).

It would be great if the index could be included :-)  IMO consistency, even in small details, can make all the difference for those using EdgeTX for the first time.

PR #4767 (2024-03-24): feat(color): show FM name and number in GV edit page
Fixes #4762 

[image]
closed 2024-03-24 by pfeerick