EdgeTX/edgetx #4765: Board: Jumper T20 Error: Unable to load file :/hwdefs/t20.json - Companion 2.10.0 RC1

Mavy1704, 2024-03-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I have the message "Board: Jumper T20 Error: Unable to load file :/hwdefs/t20.json" with the Jumper T20 and EdgeTX Companion 2.10.0 and the 'special functions only work on the T20 not in the Campanion?

### Expected Behavior

I have the message "Board: Jumper T20 Error: Unable to load file :/hwdefs/t20.json" with the Jumper T20 and EdgeTX Companion 2.10.0 and the 'special functions only work on the T20 not in the Campanion?

### Steps To Reproduce

I have the message "Board: Jumper T20 Error: Unable to load file :/hwdefs/t20.json" with the Jumper T20 and EdgeTX Companion 2.10.0 and the 'special functions only work on the T20 not in the Campanion?

### Version

Other (Please specify below)

### Transmitter

Jumper T20/T20S

### Operating System (OS)

Windows

### OS Version

Win11

### Anything else?

I have the message "Board: Jumper T20 Error: Unable to load file :/hwdefs/t20.json" with the Jumper T20 and EdgeTX Companion 2.10.0 and the 'special functions only work on the T20 not in the Campanion?

3djc, 2024-03-18:
To be expected, will be in next RC: https://github.com/EdgeTX/edgetx/issues/4700

closed 2024-03-18 by 3djc

fa-lk, 2024-05-24:
@3djc I might be doing something wrong, but I still get the same error message with 2.10 RC4 directly after installation if I create a new radio profile

pfeerick, 2024-05-24:
Why not 2.10.0? And is this T20 v2? I can't repo this for the T20, but can for the T20v2...

fa-lk, 2024-05-24:
I also tried 2.10.0.0, same error. t20 and t20v2 are not binary compatible, perhaps thats why.....no clue :)

pfeerick, 2024-05-24:
You shouldn't be getting that error for the T20, but should for the T20v2... nothing to do with binary compatibility... simulator must be missing.

fa-lk, 2024-05-24:
Okay, got it, thanks!

elecpower, 2024-05-24:
If simulator not generated prior to Companion compile there will be no hwdef file for the radio.
I get the same result as @pfeerick T20 okay and T20V2 error.

pfeerick, 2024-05-24:
#5058 should fix the T20v2 issue... I think everything was in place to use the T20 simulator UI... just libsimu/hwdef missing