EdgeTX/edgetx #4769: Flysky AFHDS2A - telemetry streaming should not be limited to just when RSSI sensor present

pfeerick, 2024-03-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

[from rcg user MachineCat](https://www.rcgroups.com/forums/showpost.php?p=52260943&postcount=6899):

> Just saw you guys talking about RSSI of AFHDS 2A. Actually I have a different question related to that.
> 
> In the `telemetry/flysky_ibus.cpp`, is looks like the code only set the "telemetry streaming" value when there is valid RSSI reading. However, a lot of my RX do not send RSSI. But they do send RX battery voltage which is what I am interested in.

### Describe the solution you'd like

> So I think we should set the "telemetry streaming" in more conditions than just RSSI.

pfeerick, 2024-03-19:
It seems the NV14 implementation instead simply checks the sensorCount to consider telemetryStreaming active... :thinking: 

https://github.com/EdgeTX/edgetx/blob/e89f77b608fa90e4a7f26d00d2f26cc4c1f160dd/radio/src/telemetry/flysky_nv14.cpp#L298C1-L300C4

rdeanchurch, 2024-03-19:
In a related, but not specific to the above:
In Logical Switches and Special Functions there is a Boolean 'Telemetry' that can be used.
I assume that is close to or identical to 'telemetryStreaming' mentioned above.

If that is true it would be nice if a pseudo sensor named 'Telm' (or something similar) was available to EdgeTx GUI and Lua API.

As it is it please note the code Offer Shmuely wrote in the 'Value2' widget (lib_widget_tools.lua) to get around not having  such an indicator,
"
function M.isTelemetryAvailable()
...
 local tele_src = getFieldInfo("RSSI")
        if not tele_src then tele_src = getFieldInfo("RxBt") end
        if not tele_src then tele_src = getFieldInfo("A1") end
        if not tele_src then tele_src = getFieldInfo("A2") end
        if not tele_src then tele_src = getFieldInfo("1RSS") end
        if not tele_src then tele_src = getFieldInfo("2RSS") end
        if not tele_src then tele_src = getFieldInfo("RQly") end
        if not tele_src then tele_src = getFieldInfo("TRSS") end
        if not tele_src then tele_src = getFieldInfo("VFR%") end
"

Did he get them all?   
Are there redundancies?
 I don't know!
But a pseudo/virtual sensor "Telm' would streamline the code.
Please consider.

rdeanchurch, 2024-03-19:
> Lua getRSSI() actually returns the status of telemetry streaming

Thank you.
That wasn't obvious to me.